org-arl/arlpy #1: Error in demo1.py

mchitre, 2017-07-18:
Error reported via email by Sérgio Miguel:

> Gave me this error when I simulated the demo1.py file that is in arlpy,
> can you explain to me what I can do in order to correct this error.
> I´m new in Python language.
> 
>   File "D:\Anaconda3\lib\site-packages\arlpy\comms.py", line 16, in <module>
>     from signal import time as _time
> 
> ImportError: cannot import name 'time'
>

mchitre, 2017-07-18:
Sérgio, this is indeed strange as the function `time` exists in the `signal.py`. Could you please check if `signal.py` exists in your `site-packages\arlpy` folder?  How did you install (from GitHub source or using `pip`)?  If from source, are you using the `master` branch?

smalhao, 2017-07-18:
Yes i have the file signal.py in site-packages\arlpy folder, i install with pip

mchitre, 2017-07-20:
Can you just try `import arlpy.signal` and `from arlpy.signal import time as _time` from a Python shell to ensure that your `site-packages\arlpy` folder is in the import path?

mchitre, 2018-01-18:
Closing, due to no follow up.

closed 2018-01-18 by mchitre