org-arl/arlpy #9: call to uwapm.models() gives error

oliskir, 2018-10-19:
Having installed the acoustic toolbox and the arlpy library, I tried to replicate the bellhop notebook, but failed at the very first step. The call to pm.models() gives me the following error:
```terminal
At line 39 of file ReadEnvironment.f90 (unit = 5, file = '/tmp/tmp61mcdwv1.env')
Fortran runtime error: End of file

Error termination. Backtrace:
#0  0x7f3665d362da in ???
#1  0x7f3665d36ec5 in ???
#2  0x7f3665d3768d in ???
#3  0x7f3665eada33 in ???
#4  0x7f3665ea69c4 in ???
#5  0x7f3665ea80f9 in ???
#6  0x55cfad042bcb in ???
#7  0x55cfad035e3c in ???
#8  0x55cfad03091e in ???
#9  0x7f3665395b96 in ???
#10  0x55cfad030979 in ???
#11  0xffffffffffffffff in ???
``` 
The problem seems to be that the bellhop.exe program tries to read the empty .env file (/tmp/tmp61mcdwv1.env), while it expects a .env with some appropriately formatted content. 

I am using Ubuntu 18, gfortran compiler, and python 3.6.5 64 bit.

mchitre, 2018-10-19:
The call to `pm.models()` runs `bellhop.exe` with an empty env file. This is the expected behavior. But bellhop isn't supposed to terminate with a runtime error. On my machine, I get this back from Bellhop:
```
$ bellhop.exe test.env 
STOP Fatal Error: See print file for details
```

So I suspect this is to do with the version of Bellhop or with the compilation options. What version of Bellhop are you using?

Also, can you try running this in a new notebook:
```
import arlpy.uwapm as pm
import arlpy.plot as plt
import numpy as np

env = pm.create_env2d()
rays = pm.compute_eigenrays(env, model='bellhop')
pm.plot_rays(rays, env=env, width=900)
```
and see if it works...? This will skip the check for bellhop, and create a env file with proper content and run bellhop on it. It'll help us debug the problem.

oliskir, 2018-10-19:
Hi mchitre, thanks for the quick answer! Yes, with a proper environment file, the code executes nicely. So the problem only occurs when the environmental file is empty. I get the same behaviour if I run bellhop.exe with an empty env file in the terminal, independent of your library.

mchitre, 2018-10-20:
OK, probably a change in bellhop.exe behavior, or differences across platforms. Let me try and test with the latest version, and I'll update the mechanism to test availability of bellhop in the next release.

In the mean time, I presume you shouldn't be blocked, as you can just add the `model = 'bellhop'` to your `uwapm` calls, and it'll work fine without needing the detection.

oliskir, 2018-10-22:
Yes, no urgency. I can continue working in the mean time. 

Unrelated to this, I was wondering if you have plans to implement 3d environments and execution of bellhop3d?

mchitre, 2018-10-22:
I haven't explored the file formats for Bellhop 3D since I don't use it. But it will be useful someday for me, I'm sure, and so I'll be happy to implement it if there's interest by others in using it. If you feel that you'll find value in it, do raise a feature request and I'll attend to it as soon as I get some time.

mchitre, 2019-07-05:
Hopefully fixed. I can't test since I don't seem to be able to replicate the issue, but it now uses a non-existent file instead of an empty file, and at least on OSX, that seems to print an error message and gracefully terminate.

closed 2019-07-05 by mchitre