org-arl/arlpy #11: Eigen rays are not correct

ZHANGGU, 2018-10-26:
Hei

Great python package!

I tried scenarios as shown by the figures, the eigen rays are not correct somehow.

1. It is wrong in env setup, as the Rx is below the bottom, but it gives all rays.

[image]

2. It should not give all rays.
[image]

By the way, using Bokeh plot is resource consuming, and it takes much time to give a figure of low resolution in saving it. While it is much faster in zoom-in/out

Thanks

ZHANGGU, 2018-10-26:
[image]

mchitre, 2018-10-26:
Thanks for the feedback.

I don't know the exact setup you used, but it looks like the environment parameters you are using is generating too many rays. You may wish to only generate eigenrays rather than all rays. In any case, please refer to Bellhop documentation on how the modeling is controlled by the parameters, as all arlpy does is wrap Bellhop, and does not do the ray computations itself.

ZHANGGU, 2018-10-26:
Hi 
Do you have detailed Bellhop manual? I can see one manual of 2011 on the web.

Thanks.

mchitre, 2018-10-26:
Yes, I believe the 2011 one is the most recent one that I know of.

ZHANGGU, 2018-10-30:
Hi, I did read the 2011 manual which is an intro really. I am still wondering at you would have the intro about output data structures. If yes, could you please share? As my simulations as shown are not correct at all. I am stucked in getting the correct rays. Appreciate your help!

mchitre, 2018-11-01:
The original acoustics toolbox has a number of MATLAB scripts to deal with the output. Those are the best source of information on the output formats for me, as they are current and work. My Python implementations were based on that, and as far as I can tell from the use cases I use it for, that seems to work fine.

closed 2018-11-01 by mchitre