org-arl/arlpy #12: Arlpy donot plot correctly

ZHANGGU, 2018-10-31:
Hei

I can see arlpy.plot_ray will not show the rays who have the maks number of bottom bounces.

Some times, I can see it has some reasons coz no returning rays will not be shown.

In my trial, I can see that the ray that has max bottom bounces is appearent correct. 

On draging the plot, the hided plot is shown. see below zooming in
[image]
[image]

mchitre, 2018-11-01:
Thanks for reporting and taking the time to annotate the plots.

However, I don't understand the issue. The 2 plots seem to be the same, other than zooming, and I don't understand what the red arrows point to. Could you share the code snippet of what you are trying to simulate, and what you expect as a plot, and I might be able to understand better.

mchitre, 2018-11-01:
Ah! I understand what you mean. The color map is such that some rays become white. I'll change the color map in the next release to fix this.

ZHANGGU, 2018-11-01:
Thanks. It is not easy :-) once you turn on plotting all rays, there comes another issue. Coz the plot func is reused for plotting all rays, too. For all rays, there are returning rays ( see the figure), and I do not know why it happens and how to void those, do you have any advice? Per today, this method will not show the returnning rays.
[image]

mchitre, 2018-11-01:
I don't understand what you mean by "returning rays". The way Bellhop calculates eigenrays is to send out a fan of rays and pick the ones that pass close to the receiver. As a result, you often get 2 closely spaced rays going on each side of the receiver. Is that way you are referring to?

Also, would help for me to see your code snippet to understand what you are simulating.

ZHANGGU, 2018-11-01:
See the picture as below, it is a ray from computation of all rays (1000m to Rx) not just eigen rays. This ray computation shows that it returns after 430 m. How to void such returning rays which exist among the all ray computations.
[image]

mchitre, 2018-11-01:
Ok, I understand what you mean by "returning rays" now. They should not exist, there's probably some bug in decoding the ray file. Can you give me a sample script that can replicate this, so I can track down the bug and squash it?

ZHANGGU, 2018-11-01:
Hei

Enclosed are env and bty files.

As talked, these returning rays come from BELLHOP computation, not your python.

I am confused, how they come and how to void those in the results

Thanks again

Regards

Guosong

ZHANGGU, 2018-11-01:
The returning rays is due to the angle of incidence and the gradient of the bottom. those returning rays shall be removed by verifications not just his simple hiding method.

mchitre, 2019-07-05:
I believe this is resolved, so closing the issue.

closed 2019-07-05 by mchitre