org-arl/arlpy #16: Possible divide by zero in plot_rays

rigelifland, 2018-11-26:
The greyscale color of the ray lines are calculated using 
```
max_amp = _np.max(_np.abs(rays.bottom_bounces))
...
c = int(255*_np.abs(row.bottom_bounces)/max_amp)
```

In the case of no bottom bounces (which isn't uncommon for eigen rays in sufficiently close-range/deep environments), this results in a divide by zero error.

Possible fix:
```
#Get the range, using one less than the minimum number of bottom bounces so that the minimum doesn't plot white.
amp_range = [_np.min(_np.abs(rays.bottom_bounces)) - 1, _np.max(_np.abs(rays.bottom_bounces))]
...
c = int(_np.interp(_np.abs(row.bottom_bounces), amp_range, [0, 255])) 
```

[Link to relevent code](https://github.com/org-arl/arlpy/blob/213f5c7b48c76866e4b275259c91bc8c439fca09/arlpy/uwapm.py#L447)

mchitre, 2018-11-27:
Thanks! Will fix in the upcoming release.

mchitre, 2019-07-05:
Fixed.

closed 2019-07-05 by mchitre