org-arl/arlpy #18: ValueError: could not convert string to float: "'2D'" in bellhop.ipynb

elishafer, 2018-12-15:
Hi,

In the ipython bellhop sample I'm getting the following error:

```

arrivals = pm.compute_arrivals(env)
pm.plot_arrivals(arrivals, width=900)
arrivals = pm.compute_arrivals(env)
pm.plot_arrivals(arrivals, width=900)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-6-3843dd94e0a3> in <module>
----> 1 arrivals = pm.compute_arrivals(env)
      2 pm.plot_arrivals(arrivals, width=900)

~/.virtualenvs/dsp3/lib/python3.5/site-packages/arlpy/uwapm.py in compute_arrivals(env, model, debug)
    292     if debug:
    293         print('[DEBUG] Model: '+model_name)
--> 294     return model.run(env, arrivals, debug)
    295 
    296 def compute_eigenrays(env, tx_depth_ndx=0, rx_depth_ndx=0, rx_range_ndx=0, model=None, debug=False):

~/.virtualenvs/dsp3/lib/python3.5/site-packages/arlpy/uwapm.py in run(self, env, task, debug)
    563         fname_base = self._create_env_file(env, taskmap[task][0])
    564         if self._bellhop(fname_base):
--> 565             results = taskmap[task][1](fname_base)
    566         else:
    567             results = None

~/.virtualenvs/dsp3/lib/python3.5/site-packages/arlpy/uwapm.py in _load_arrivals(self, fname_base)
    669     def _load_arrivals(self, fname_base):
    670         with open(fname_base+'.arr', 'rt') as f:
--> 671             freq, tx_depth_count, rx_depth_count, rx_range_count = self._readf(f, (float, int, int, int))
    672             tx_depth = self._readf(f, (float,)*tx_depth_count)
    673             rx_depth = self._readf(f, (float,)*rx_depth_count)

~/.virtualenvs/dsp3/lib/python3.5/site-packages/arlpy/uwapm.py in _readf(self, f, types)
    664         for j in range(len(p)):
    665             if len(types) > j:
--> 666                 p[j] = types[j](p[j])
    667         return tuple(p)
    668 

ValueError: could not convert string to float: "'2D'"
```

Using Python version 3.5.2 on virtual env.

I suspect that this may be because of python version. Could you please state the version of Python that you're using?

Thanks!

elishafer, 2018-12-15:
Update, same issue on Python 3.6.

mchitre, 2018-12-15:
I'm using Python 3.6.4, but I doubt the problem is due to Python version. I suspect it has something to do with bellhop version or execution.

Could you try `pm.compute_arrivals(env, debug=True)`? This will leave the Bellhop files around and print their location. If you can zip them up and attach them, that would help debugging. It's the arrival file (`*.arr`) that seems to be the problem, but it may be useful to have all the files anyway.

elishafer, 2018-12-15:
`pm.compute_arrivals(env, debug=True)` gives the same error.
I did compile the bellhop.exe on my machine without installing any extra tools. The source I downloaded from the [ocean acoustics toolbox](http://oalib.hlsresearch.com/Modes/AcousticsToolbox/).
I'm new to this library so I don't know where the `.arr` file is located. Could you point me in the right direction?

elishafer, 2018-12-15:
OK, I see that the `.arr` file is in the `/tmp/` folder. 
I'm also getting a message in the Jupyter notebook CLI terminal:
```
At line 39 of file ReadEnvironment.f90 (unit = 5, file = '/tmp/tmpcv_v2w_r.env')
Fortran runtime error: End of file
```
[prt_env_arr_files.zip](https://github.com/org-arl/arlpy/files/2682988/prt_env_arr_files.zip)

mchitre, 2018-12-15:
Thanks! That helped identify the problem!!

The file format for the `arr` file seems to have changed in the latest Bellhop. I just downloaded and recompiled the latest Bellhop and was able to reproduce the problem.

Old version:
```
   25000.000000000000                1           1           1
   5.00000000    
   10.0000000    
   1000.00000    
          34
          34
   1.25386987E-05   1636.07935      0.721795559      -4.85707142E-06  -22.5382519       22.5382519               9           8
  :
  :
```

New version:
```
 '2D'
   25000.000000000000     
           1   5.00000000    
           1   10.0000000    
           1   1000.00000    
          34
          34
   1.25360748E-05   1636.0794715189722       0.721795619      -4.85707187E-06  -22.5382576       22.5382576               9           8
  :
  :
```

The change is small, and only in the headers, and so should be easy to fix. But I need to find a way to introduce it so that it doesn't break compatibility with the older versions, as many users are likely to have those installed. Give me a day or two to sort this out in a robust way, and I'll release a version for you to try.

elishafer, 2018-12-15:
Cool, thanks.
I downloaded an older version of the bellhop [here](https://pulsar.byu.edu/dfvankom/acoustics-toolbox) and it's working now.

mchitre, 2019-07-05:
Fixed in `dev`. Will be part of the upcoming release.

closed 2019-07-05 by mchitre