org-arl/arlpy #22: Possibility to change the number of output beams from the source in uwapm

cristiantesconi1994, 2019-03-11:
Hello,
I'm using arlpy library. I desired to change the number of output beams from the source.
For example, if i want to visualize only 2 or 3 rays when i run plot_ray routine, how can i do that?
I notice that in the routine create_env2d there isn't any parameters who permits to modify the number of beams. 
I desired, to modify as in Matlab in the .env file, the number of beam.
Waiting for an answer I send my best regards.

Cristian Tesconi

cristiantesconi1994, 2019-03-12:
Yes, I refer to the NBEAMS parameter in the env file. If it possible, i'd like it available. I await news.
Thanks in advance.

mchitre, 2019-03-12:
Okay, I have added it to the current `dev` branch in commit ff416e5efee6f79d27576ffdf73671593af13c6b. It should be part of the upcoming release, but if you need it now, you can simply work from the current `dev` branch as it is quite stable.

cristiantesconi1994, 2019-03-12:
Ok thanks. I have another question: i'm using Atom and Jupyter notebook, how can i change the branch and switch from master to dev ?

mchitre, 2019-03-12:
You can clone the repo: `git clone git@github.com:org-arl/arlpy.git`, switch to the `dev` branch: `git checkout dev`, and then set your `PYTHONPATH` to point to the directory where you've cloned the repo. Delete off the previous `arlpy` version from your machine/path to ensure that there's no clash between versions.

cristiantesconi1994, 2019-03-12:
Ok, thank you very much for the availability.

closed 2019-04-03 by mchitre