org-arl/arlpy #23: Problem with Bellhop version

cristiantesconi1994, 2019-03-12:
Hi,
I'm testing the software, but i have some problem. The problem i am exposing now has already been treated in issue #21: i tried to use the latest version of bellhop (bellhop.exe) and i obtain the sequent error: 
[image], but plotray and eigenray work better.
So i tried to use previus version of bellop and the error described above no longer occurs, but plot_ray and eigenray don't work, obtaining the following error: 
[image].
Can you give me any suggestion? how can i proceed?

Thanks very much in advance.

mchitre, 2019-03-13:
I'm surprised there is a problem with older version of `bellhop` and plotting rays, since many regularly use that. Anyway, its time that `arlpy` moved to the latest `bellhop` and so I'll get that done as soon as possible.

cristiantesconi1994, 2019-03-13:
I downloaded Bellop from http://oalib.hlsresearch.com/AcousticsToolbox/  and i copied the Bellhop.exe in the directory where i have my python script. Firstly i downloaded from the usual link above the Bellhop version in: 
[image]
and then i tried to use bellhop version in:
[image]
Have i tried correct version?
if not, could you attach bellhop.exe to put in the directory where I have my python script?

Thanks in advance.

mchitre, 2019-07-05:
The latest `dev` should have fixed the issue with the latest `bellhop.exe` version. This will be part of the upcoming release.

closed 2019-07-05 by mchitre