org-arl/arlpy #26: call to uwapm.models() gives error - continues to be an issue under Windows/10 x64 with Anaconda Python 3.6.7 and arlpy 1.6

U-S-S, 2019-07-09:
Believing specified environment may have been to complex, replaced bathymetry and ssp with a simple set-up.  Results remain the same - receive error related to missing .ray file (in jupyter-notebook window) and mis-formed env file (in Win cli).
Have attached Jupyter notebook and resultant .env/.prt/.bty file(s) for examination and debugging.

Regards

USS 
Platform info.......

# Name                    Version                   Build  Channel
arlpy                     1.6                      pypi_0    pypi

================
     active environment : arlpy
...
               platform : win-64
             user-agent : conda/4.7.5 requests/2.21.0 CPython/3.7.1 Windows/10 Windows/10.0.17763
          administrator : False
             netrc file : None
           offline mode : False
==========================================

[arlpy-uwapm-test.9Jul2019pdf.pdf](https://github.com/org-arl/arlpy/files/3374927/arlpy-uwapm-test.9Jul2019pdf.pdf)
[arlpy-uwapm-test.9Jul2019CLI-error_output.txt](https://github.com/org-arl/arlpy/files/3374928/arlpy-uwapm-test.9Jul2019CLI-error_output.txt)
[arlpy-uwapm-test.9Jul2019_PRT_ENV_BTY_files.txt](https://github.com/org-arl/arlpy/files/3374945/arlpy-uwapm-test.9Jul2019_PRT_ENV_BTY_files.txt)

mchitre, 2019-07-10:
@U-S-S thanks for the detailed report. Few questions and comments:

1. What version of Bellhop are you using?
2. From the files you attached, the problem does not seem to be in `uwapm.models()` from the files you included, but with a `uwapm.compute_eigenrays()`. Or are you reporting two different errors?
3. The problem with your setup seems to be the `ssp`. I was able to reproduce your error and fix it. Change the bottom-most `ssp` entry to `405`  (see below) and it seems to work fine. Bellhop doesn't seem to like you giving sound speed below the lowest point in your bathymetry.
```
ssp = [
  [0,1540],
  [90, 1530],
  [200, 1532],
  [300, 1533],
  [405, 1535]
]
```

U-S-S, 2019-07-10:
@mchitre

Many thanks for setting me on the right course - got it working!
In answer to your questions:
1. am using the WIN 10 x64 Bellhop binary from http://oalib.hlsresearch.com/AcousticsToolbox/ which I believe is version 2018_7
2. Sorry for the confusion only the one issue, which you are correct in indicating likely is a matter related to I/O to uwapm.models()
3. Your correction worked by setting the last SSP profile depth = depth of the lowest depth in the bathymetry range.  I was mislead to extrapolating the SSP beyond the bottom by an error I received in an earlier trial run which indicated 'Last depth in soundspeed array must be beyond water depth: ...' which suggested extrapolation to beyond the bottom.

Again many thanks!

Any thoughts of extending the arlpy capability to building Range-dependent Sound Speed Proﬁle Inputs as per http://oalib.hlsresearch.com/Rays/HLS-2010-1.pdf?  There are now numerous 4D ocean model data sources which could feed such an input.

Regards,

U-S-S

closed 2019-07-10 by U-S-S

mchitre, 2019-07-12:
Thanks U-S-S, I've raised a feature request for the Range-dependent Sound Speed Proﬁle feature in `arlpy.uwapm`.