org-arl/arlpy #31: Bokeh - Deprecation warning and Bad 'legend' parameter value:

patel999jay, 2019-11-09:
I had installed arlpy with new version with pip install arlpy and pip also installed newer version of bokeh 1.3.4 but unfortunately arlpy didn't work with newer version of Bokeh and throws you Deprecation warning and ValueError: Bad 'legend' parameter value: None while plotting.

`BokehDeprecationWarning: 'legend' keyword is deprecated, use explicit 'legend_label', 'legend_field', or 'legend_group' keywords instead
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\jay_p\AppData\Local\Programs\Python\Python37-32\lib\site-packages\arlpy\uwapm.py", line 225, in plot_env
    _plt.plot([min_x, max_x], [0, 0], xlabel=xlabel, ylabel='Depth (m)', xlim=(min_x-mgn_x, max_x+mgn_x), ylim=(-max_y-mgn_y, -min_y+mgn_y), color=surface_color, **kwargs)
  File "C:\Users\jay_p\AppData\Local\Programs\Python\Python37-32\lib\site-packages\arlpy\plot.py", line 396, in plot
    _figure.line(x, y, line_color=color, line_dash=style, line_width=thickness, legend=legend)
  File "fakesource", line 5, in line
  File "C:\Users\jay_p\AppData\Local\Programs\Python\Python37-32\lib\site-packages\bokeh\plotting\helpers.py", line 932, in func
    _update_legend(self, legend_kwarg, glyph_renderer)
  File "C:\Users\jay_p\AppData\Local\Programs\Python\Python37-32\lib\site-packages\bokeh\plotting\helpers.py", line 489, in _update_legend
    _LEGEND_KWARG_HANDLERS[kwarg](value, legend, glyph_renderer)
  File "C:\Users\jay_p\AppData\Local\Programs\Python\Python37-32\lib\site-packages\bokeh\plotting\helpers.py", line 416, in _handle_legend_deprecated
    raise ValueError("Bad 'legend' parameter value: %s" % label)
ValueError: Bad 'legend' parameter value: None`

You can solve this error with degrade the version of Bokeh 1.3, Please install manually Bokeh 1.3 by following command :

`python -m pip install bokeh==1.3.0`

if you don't know your current version of bokeh then please check with following command : 
`python -m pip freeze`

mchitre, 2019-11-09:
Noted. Will update to be compliant with latest bokeh in next release.

patel999jay, 2019-11-20:
Is there any way we can combine this tools with NS-3 ?

mchitre, 2019-12-01:
I'm not sure I understand what you mean by combine with NS-3. Could you explain?

mchitre, 2019-12-01:
Deprecation warnings are fixed in latest `dev`. Slated to be part of the next release (`1.6.2`).

closed 2019-12-01 by mchitre

patel999jay, 2019-12-02:
> I'm not sure I understand what you mean by combine with NS-3. Could you explain?

Can you combine this tool with [Network Simulator-3](https://www.nsnam.org/docs/release/3.9/doxygen/group___u_a_n.html) to add new/configure channel parameters  ? Is there any way we can use this bellhop returns to pass to NS-3 ?

mchitre, 2019-12-03:
@patel999jay I've not explored what will be needed to integrate with NS-3. However, if you want to do underwater network simulations, there are several options available:
- [UnetStack](www.unetstack.net) is the one that I use. This is a Java/Groovy based simulator that allows custom channel models. Integrating Bellhop to this should be quite easy.
- [DESERT](http://desert-underwater.dei.unipd.it) is one that is based on NS, and I recall that they had some form of integration with Bellhop through [WOSS](http://telecom.dei.unipd.it/ns/woss/). I have not used it, but you may perhaps find what you need in this ecosystem.