org-arl/arlpy #34: Documentation may need an update

harivnkochi, 2019-12-14:
On the documentation page (https://arlpy.readthedocs.io/en/latest/uwapm.html), this is an example code given: 

```
 import arlpy.uwapm as pm
 env = pm.create_env2d(depth=20, soundspeed=[[0,1540], [5,1535], [20,1530]])
```

However, when I run this code, I get the error

> ValueError: ('soundspeed profile must have at least 4 points',)

Does the documentation need to be updated ?

harivnkochi, 2019-12-14:
the link to acoustics toolbox on the documentation page for uwapm (https://arlpy.readthedocs.io/en/latest/uwapm.html) and on the sample notebook (https://arlpy.readthedocs.io/en/latest/_static/bellhop.html) is obsolete.  The new link is probably http://oalib.hlsresearch.com/AcousticsToolbox/

mchitre, 2019-12-20:
Thanks, will fix.

closed 2020-04-23 by mchitre