org-arl/arlpy #37: uwapm seems broken after bokeh upgrade to 1.4

harivnkochi, 2020-01-27:
I just upgraded bokeh to version 1.4, and I'm getting errors using uwapm (specifically, the createenv2d() function). 

When I revert back to bokeh version 1.3, the error goes away.
I get the following warning

`BokehDeprecationWarning: 'legend' keyword is deprecated, use explicit 'legend_label', 'legend_field', or 'legend_group' keywords instead`

followed by the error:

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-9-41565c6176c8> in <module>
     13     nbeams=2500 #at 2500, u see an extra surface-reflected ray of different type
     14 )
---> 15 pm.plot_ssp(env)

~/opt/anaconda3/lib/python3.7/site-packages/arlpy/uwapm.py in plot_ssp(env, **kwargs)
    270         tck = _interp.splrep(s[:,0], s[:,1], s=0)
    271         xnew = _interp.splev(ynew, tck, der=0)
--> 272         _plt.plot(xnew, -ynew, xlabel='Soundspeed (m/s)', ylabel='Depth (m)', hold=True, **kwargs)
    273         _plt.plot(s[:,1], -s[:,0], marker='.', style=None, **kwargs)
    274     else:

~/opt/anaconda3/lib/python3.7/site-packages/arlpy/plot.py in plot(x, y, fs, maxpts, pooling, color, style, thickness, marker, filled, size, mskip, title, xlabel, ylabel, xlim, ylim, xtype, ytype, width, height, legend, hold, interactive)
    394         _figure.add_layout(_bmodels.Label(x=5, y=5, x_units='screen', y_units='screen', text=desc, text_font_size="8pt", text_alpha=0.5))
    395     if style is not None:
--> 396         _figure.line(x, y, line_color=color, line_dash=style, line_width=thickness, legend=legend)
    397     if marker is not None:
    398         scatter(x[::(mskip+1)], y[::(mskip+1)], marker=marker, filled=filled, size=size, color=color, legend=legend, hold=True)

fakesource in line(self, x, y, **kwargs)

~/opt/anaconda3/lib/python3.7/site-packages/bokeh/plotting/helpers.py in func(self, **kwargs)
    930 
    931         if legend_kwarg:
--> 932             _update_legend(self, legend_kwarg, glyph_renderer)
    933 
    934         self.renderers.append(glyph_renderer)

~/opt/anaconda3/lib/python3.7/site-packages/bokeh/plotting/helpers.py in _update_legend(plot, legend_kwarg, glyph_renderer)
    487     kwarg, value = list(legend_kwarg.items())[0]
    488 
--> 489     _LEGEND_KWARG_HANDLERS[kwarg](value, legend, glyph_renderer)
    490 
    491 

~/opt/anaconda3/lib/python3.7/site-packages/bokeh/plotting/helpers.py in _handle_legend_deprecated(label, legend, glyph_renderer)
    414 
    415     if not isinstance(label, (string_types, dict)):
--> 416         raise ValueError("Bad 'legend' parameter value: %s" % label)
    417 
    418     if isinstance(label, dict):

ValueError: Bad 'legend' parameter value: None
```

mchitre, 2020-02-22:
Ah yes, bokeh changed the syntax. I will update in the upcoming release.

rich9403, 2020-04-22:
Line 398 also has the syntax "legend".  It did not appear that you corrected this line in:
[https://github.com/org-arl/arlpy/commit/6d1734cf5fd1439351a8221e57da5a8d6dfba49c](url)

mchitre, 2020-04-23:
@rich9403 thanks for the heads up. This issue isn't "resolved" yet, so there will be places which need fixing. It's tagged to be addressed in the upcoming release.

mchitre, 2020-04-23:
@rich9403 just checked ... there doesn't seem to be a "legend" called on Bokeh anymore. The "legend" syntax for `arlpy` is still valid, and you pointed to a line where it was called on `scatter`? That is a `arlpy` function is supports option `legend`.

closed 2020-04-23 by mchitre