org-arl/arlpy #38: uwapm.compute_arrivals(env) doesnt work if multiple very close receiver ranges are specified

harivnkochi, 2020-01-27:
Looks like there is a minimum range resolution at which the receivers should be placed in order to avoid this error. It might be good to be able to catch/pre-empt this error.

Example code and error:
```
env = pm.create_env2d(
    frequency=2000,
    depth= 150,
    soundspeed= 1540,
    bottom_soundspeed=1600,
    bottom_density=1200,
    bottom_absorption=10.0,
    tx_depth=tx_depth,
    rx_depth= np.arange(1,5,1),#
    rx_range= np.arange(10,10.1,.01), 
    min_angle=-45,
    max_angle=45,
    nbeams=2500 
)
arrivals = pm.compute_arrivals(env)

---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-43-19a6d64bab5c> in <module>
----> 1 arrivals = pm.compute_arrivals(env)

~/opt/anaconda3/lib/python3.7/site-packages/arlpy/uwapm.py in compute_arrivals(env, model, debug)
    293     if debug:
    294         print('[DEBUG] Model: '+model_name)
--> 295     return model.run(env, arrivals, debug)
    296 
    297 def compute_eigenrays(env, tx_depth_ndx=0, rx_depth_ndx=0, rx_range_ndx=0, model=None, debug=False):

~/opt/anaconda3/lib/python3.7/site-packages/arlpy/uwapm.py in run(self, env, task, debug)
    569         fname_base = self._create_env_file(env, taskmap[task][0])
    570         if self._bellhop(fname_base):
--> 571             results = taskmap[task][1](fname_base)
    572         else:
    573             results = None

~/opt/anaconda3/lib/python3.7/site-packages/arlpy/uwapm.py in _load_arrivals(self, fname_base)
    679 
    680     def _load_arrivals(self, fname_base):
--> 681         with open(fname_base+'.arr', 'rt') as f:
    682             hdr = f.readline()
    683             if hdr.find('2D') >= 0:

FileNotFoundError: [Errno 2] No such file or directory: '/var/folders/s4/0jrt39zs65j81br7ll9g8nfc0000gn/T/tmpen87jupi.arr'
```

mchitre, 2020-02-22:
Perhaps. But it would be impossible to catch all the things that Bellhop can't handle. But what I can do is perhaps give a better error message.

mchitre, 2020-04-23:
Likely same problem as #39. I'll increase decimal places to allow up to cm accuracy.

mchitre, 2020-04-23:
Improved error reporting:
```
[WARN] Bellhop did not generate expected output file

[BELLHOP]  *** FATAL ERROR ***
[BELLHOP]  Generated by program or subroutine: SdRdRMod
[BELLHOP]  Receiver ranges are not monotonically increasing
[BELLHOP] 
```

closed 2020-04-23 by mchitre