org-arl/arlpy #39: parsing custom env failed to recreate environment

patel999jay, 2020-02-19:
I parsed custom env to module but it failed to regenerate simulation scenarios. I had successfully tested the same scenario files with bellhop using matlab, works perfectly fine. Also is there any way to parse non-monotonic ssp ? I am using Bathmetry from [GEBCO 2019](https://www.gebco.net/data_and_products/gridded_bathymetry_data/). Any help would be appreciated.

mchitre, 2020-02-22:
Can you post the `arlpy` code that you used?

patel999jay, 2020-02-22:
Here is the code that i used,

```bash
import arlpy.uwapm as pm
import arlpy.plot as plt
import numpy as np

env = pm.create_env2d()
pm.print_env(env) # to check env

ssp = [
           [0,         0.0022477890249762906],
           [1446.0123291015625,            1452.5137939453125],
           [1463.50646209716797,          1473.13202667236328],
           [1466.33708953857422,           1477.57839965820312],
           [1469.17899703979492,           1480.42838668823242],
           [1472.03192138671875,           1483.28749084472656],
           [1474.89564895629883,           1486.15559768676758],
           [1477.76937103271484,           1489.03192901611328],
           [1480.65291595458984,           1491.91645812988281],
           [1483.54705810546875,           1494.80963134765625],
           [1486.45201873779297,           1497.71190643310547],
           [1489.36491394042969,           1500.62077331542969],
           [1492.28920745849609,           1503.53911590576172],
           [1495.223388671875,           1506.46572875976562],
           [1498.17044067382812,           1509.40312957763672],
           [1501.11749267578125,           1512.34053039550781],
           [1504.07640838623047,           1515.28792572021484],
           [1507.049560546875,             1518.247314453125],
           [1510.02271270751953,           1521.20670318603516],
           [1513.00946044921875,           1524.17765045166016],
           [1516.00753784179688,           1527.15822982788086],
           [1519.005615234375,           1530.13880920410156],
           [1522.01885986328125,            1533.1323127746582],
           [1525.040771484375,           1536.13320159912109],
           [1528.06268310546875,           1539.13409042358398],
           [1531.10107421875,           1542.14926147460938],
           [1534.14564514160156,            1545.1697883605957],
           [1537.19021606445312,           1548.19031524658203],
           [1540.25243377685547,           1551.22611999511719],
           [1543.31857299804688,           1554.26531982421875],
           [1546.38471221923828,           1557.30451965332031],
           [1549.46927642822266,           1560.36004638671875],
           [1552.55568313598633,           1563.41720581054688],
           [1555.64208984375,             1566.474365234375],
           [5101,                       1566.474365234375]
]
bathy = [
          [0,5100],             
          [0.213841,           270.083],
          [0.31033,          270.753],
          [0.654562,           271.326],
          [1.01966,           271.259],
          [1.09659,           271.734],
          [1.53731,           271.909],
          [1.72898,           272.099],
          [1.97803,           272.138],
          [2.41875,           273.279],
          [2.43831,           274.131],
          [2.85947,           275.925],
          [3.14764,           276.604],
          [3.3015,            276.786],
          [3.74222,           276.807],
          [3.85566,           276.902],
          [3.91173,           277.162],
          [1000,              277.165]
]
env = pm.create_env2d(
    depth=bathy,
    soundspeed=ssp,
    bottom_soundspeed=1450,
    bottom_density=1200,
    bottom_absorption=1.0,
    tx_depth=50
)
pm.print_env(env) # To check new bathy

pm.plot_env(env, width=900) #to plot env
pm.plot_ssp(env) #to plot env

#I think code runs fine till here, not calculating rays

rays = pm.compute_eigenrays(env) #computer eigenrays
pm.plot_rays(rays, env=env, width=900)

rays = pm.compute_rays(env) #compute rays
pm.plot_rays(rays, env=env, width=900)
```

```bash
#complains about file missing

---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-128-dbc7cd561bf0> in <module>
----> 1 rays = pm.compute_rays(env)
      2 pm.plot_rays(rays, env=env, width=900)

~/my_project_arlpy/my_project_env/lib/python3.6/site-packages/arlpy/uwapm.py in compute_rays(env, tx_depth_ndx, model, debug)
    345     if debug:
    346         print('[DEBUG] Model: '+model_name)
--> 347     return model.run(env, rays, debug)
    348 
    349 def compute_transmission_loss(env, tx_depth_ndx=0, mode=coherent, model=None, debug=False):

~/my_project_arlpy/my_project_env/lib/python3.6/site-packages/arlpy/uwapm.py in run(self, env, task, debug)
    569         fname_base = self._create_env_file(env, taskmap[task][0])
    570         if self._bellhop(fname_base):
--> 571             results = taskmap[task][1](fname_base)
    572         else:
    573             results = None

~/my_project_arlpy/my_project_env/lib/python3.6/site-packages/arlpy/uwapm.py in _load_rays(self, fname_base)
    726 
    727     def _load_rays(self, fname_base):
--> 728         with open(fname_base+'.ray', 'rt') as f:
    729             f.readline()
    730             f.readline()

FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpjln3_ui3.ray'
```
#I also get plots for env and SSP.
[image]
[image]

Do i miss something ? Thanks in advance for your help.

mchitre, 2020-03-20:
Thanks. This helps. Will try to reproduce and fix it.

harivnkochi, 2020-04-23:
It could be the same as this issue: https://github.com/org-arl/arlpy/issues/40

mchitre, 2020-04-23:
Just tested this ... the problem is related to the resolution at which the bathymetry is specified. When rounded off to 4 decimal places in km, 2.41875 and 2.43831 become 0.0024, and Bellhop doesn't like repeated points.

This could be fixed by changing the `%0.4f` in uwapm to increase number of decimal places. But I don't see the point. The simulation doesn't have the fidelity to deal with such small variations, so my suggested fix for @patel999jay is to provide the bathymetry at a courser resolution (no finer than 0.1 m spacing).

closed 2020-04-23 by mchitre

mchitre, 2020-04-23:
Changing to 6 decimal places in latest release to avoid this problem.