org-arl/arlpy #40: uwapm: Error if last depth point of the SSP does not coincide exactly with the maximum depth in the bathymetry

harivnkochi, 2020-03-24:
The modeling throws an error if the last depth point of the SSP does not coincide exactly with the maximum depth in the bathymetry. i.e, even if the depth specified in the SSP goes beyond the maximum depth.

This could be a fixed by interpolating the SSP to include the max. depth in the bathymetry, to suppress the error.

mchitre, 2020-03-24:
Likely due to Bellhop not handling this, in which case, arlpy can't do anything about it. But will cross check and confirm.

harivnkochi, 2020-03-24:
It is indeed a Bellhop issue. Quoting the Bellhop user manual here: “The last value in SSP must correspond to the max water depth.”
[image]

mchitre, 2020-03-24:
In which case, not much to be done. Thanks.

closed 2020-03-24 by mchitre