org-arl/arlpy #43: Feature request: Table of angle of interactions of each raypath with bottom

harivnkochi, 2020-04-20:
Currently, compute_eigenrays(), and compute_arrivals() compute_rays() returns dataframes containing information on raypaths with number of surface/bottom bounces, angle of launch, angle of arrival, etc.

It would also be useful to know the angle(s) of incidence of each raypath with the bottom. This can then help us determine which rays went through supercritical/subcritical interactions. 

If we look at the ray information from Bellhop with info on the path, we may be able to pull this out by simply looking at the angles of the raypath before and after it interacts with the bottom (i.e angle of incidence and reflection).

mchitre, 2020-04-21:
Will check if this info is in any of the Bellhop output files. If available, I can pull it out.

mchitre, 2020-04-23:
Bellhop does not provide this information in the ray file. Ray file only contains angle of departure, number of surface bounces, and number of bottom bounces. We pick up all three in arlpy.

closed 2020-04-23 by mchitre