org-arl/arlpy #44: uwapm error: No such file or directory

jjeffrey, 2020-04-21:
I am getting a `No such file or directory` error for any compute call using the `arlpy.uwapm` package.

For example, even following the default Bellhop example:

```
import arlpy.uwapm as pm

env = pm.create_env2d()
arrivals = pm.compute_arrivals(env)
```
Here is the error I get:
```
---------------------------------------------------------------------------

FileNotFoundError                         Traceback (most recent call last)

<ipython-input-5-c9e594c6a409> in <module>
      2 
      3 env = pm.create_env2d()
----> 4 arrivals = pm.compute_arrivals(env)
      5 

~/anaconda3/envs/spring2019/lib/python3.7/site-packages/arlpy/uwapm.py in compute_arrivals(env, model, debug)
    293     if debug:
    294         print('[DEBUG] Model: '+model_name)
--> 295     return model.run(env, arrivals, debug)
    296 
    297 def compute_eigenrays(env, tx_depth_ndx=0, rx_depth_ndx=0, rx_range_ndx=0, model=None, debug=False):

~/anaconda3/envs/spring2019/lib/python3.7/site-packages/arlpy/uwapm.py in run(self, env, task, debug)
    569         fname_base = self._create_env_file(env, taskmap[task][0])
    570         if self._bellhop(fname_base):
--> 571             results = taskmap[task][1](fname_base)
    572         else:
    573             results = None

~/anaconda3/envs/spring2019/lib/python3.7/site-packages/arlpy/uwapm.py in _load_arrivals(self, fname_base)
    679 
    680     def _load_arrivals(self, fname_base):
--> 681         with open(fname_base+'.arr', 'rt') as f:
    682             hdr = f.readline()
    683             if hdr.find('2D') >= 0:

FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpwande4tb.arr'
```

mchitre, 2020-04-21:
@jjeffrey I suspect your bellhop install isn't working properly. Have you tried running Bellhop from the command line to see that it is in path and executes correctly?

jjeffrey, 2020-04-21:
When running `bellhop.exe` on command line, I am getting the error:
```
bellhop.exe: error while loading shared libraries: libgfortran.so.4: cannot open shared object file: No such file or directory
```
I've installed gfortran already, so I am unsure of why this error is popping up. Any thoughts?

jjeffrey, 2020-04-21:
I solved the issue by installing the `libgfortran4` package via `apt`.

closed 2020-04-21 by jjeffrey