org-arl/arlpy #45: failed to calculate eigen rays with newer version of arlpy (ubuntu 20.04)

patel999jay, 2020-04-24:
Newer version is failed to calculate ray tracing. 

Here is the output of my `pip freeze` - 
[pip_freeze.txt](https://github.com/org-arl/arlpy/files/4526677/pip_freeze.txt)

Here is the error while computing eigen rays. I had also attached copy of example notebook for reference. 

[bellhop-Copy1.pdf](https://github.com/org-arl/arlpy/files/4526689/bellhop-Copy1.pdf)

 
```bash
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-25-1af3b30f110e> in <module>
----> 1 rays = pm.compute_eigenrays(env)
      2 pm.plot_rays(rays, env=env, width=900)

~/.local/lib/python3.8/site-packages/arlpy/uwapm.py in compute_eigenrays(env, tx_depth_ndx, rx_depth_ndx, rx_range_ndx, model, debug)
    352     if debug:
    353         print('[DEBUG] Model: '+model_name)
--> 354     return model.run(env, eigenrays, debug)
    355 
    356 def compute_rays(env, tx_depth_ndx=0, model=None, debug=False):

~/.local/lib/python3.8/site-packages/arlpy/uwapm.py in run(self, env, task, debug)
    605             else:
    606                 try:
--> 607                     results = taskmap[task][1](fname_base)
    608                 except FileNotFoundError:
    609                     print('[WARN] Bellhop did not generate expected output file')

~/.local/lib/python3.8/site-packages/arlpy/uwapm.py in _load_rays(self, fname_base)
    819                     'ray': [ray]
    820                 }))
--> 821         return _pd.concat(rays)
    822 
    823     def _load_shd(self, fname_base):

~/.local/lib/python3.8/site-packages/pandas/core/reshape/concat.py in concat(objs, axis, join, ignore_index, keys, levels, names, verify_integrity, sort, copy)
    269     ValueError: Indexes have overlapping values: ['a']
    270     """
--> 271     op = _Concatenator(
    272         objs,
    273         axis=axis,

~/.local/lib/python3.8/site-packages/pandas/core/reshape/concat.py in __init__(self, objs, axis, join, keys, levels, names, ignore_index, verify_integrity, copy, sort)
    327 
    328         if len(objs) == 0:
--> 329             raise ValueError("No objects to concatenate")
    330 
    331         if keys is None:

ValueError: No objects to concatenate

```

mchitre, 2020-04-24:
@patel999jay thanks for the detailed report. I'm unable to reproduce the error, as when I run the code you have in the notebook, I get correct results.

To debug further, I suggest we look at the Bellhop files you get. Run this:
```
import arlpy.uwapm as pm
env = pm.create_env2d()
rays = pm.compute_eigenrays(env, debug=True)
```
You should get an output like this (with different filenames):
```
[DEBUG] Model: bellhop
[DEBUG] Bellhop working files: /var/folders/d5/wwnqgqpj13s503lrdb4pp0680000gn/T/tmpvj7gmy3o.*
```
In the specified folder, you should see a few relevant files, e.g.,
```
tmpvj7gmy3o.env
tmpvj7gmy3o.prt
tmpvj7gmy3o.ray
```
Zip those files and upload here for me to take a look. Thanks.

patel999jay, 2020-04-24:
Hi Dr Mandar,
I would suspect that this is the debian related issue, something might not looks right to me. It still the same, I am not able to compute eigenrays with arlpy somehow. I had tested manually run munk tests with bellhop and it did worked, although i found unusual print files for them.
[MunkB_ray.zip](https://github.com/org-arl/arlpy/files/4526842/MunkB_ray.zip)

mchitre, 2020-04-24:
@patel999jay the files in the zip are your manually created ones. In fact if I try to run it, I get a fatal error "Unknown option for SSP approximation". I'm guessing we have a very different version of Bellhop? Are you on the latest one, compiled from sources?

patel999jay, 2020-04-25:
@mchitre I had tested again your version as well. If you look at the manual bellhop run in the zip file it has some extra errors generated by program itself.
Here : 
```bash
Tracing beam      41     20.00

 *** WARNING ***
 Generated by program or subroutine: TraceRay2D
 Insufficient storage for ray trajectory

 CPU Time =        21.3    s
```
I am using ubuntu 20.04 LTS (I had tested the same in virtual system LTS 18.04, it works fine. I mean bellhop without any errors.)

mchitre, 2020-04-25:
So do I understand correctly that this is a Bellhop problem on your system, and not an issue with arlpy?

patel999jay, 2020-04-25:
Your version of bellhop compiled correctly on newer version of ubuntu. It should perfectly run the arlpy right as i can see bellhop in my path via arlpy. Does arlpy have any dependencies on any python 2 packages ?

mchitre, 2020-04-25:
No, Python 3 only

closed 2020-04-25 by mchitre

vinz-uts, 2023-10-17:
Hi, I'm trying to use Bellhop model and arlpy library and I get a similar problem. If I try to run
```python
>>> import arlpy.uwapm as pm
>>> env = pm.create_env2d()
>>> rays = pm.compute_eigenrays(env, debug=True)
[DEBUG] Model: bellhop
[WARN] Bellhop did not generate expected output file
[DEBUG] Bellhop working files: /tmp/tmp6cp_l3tx.*
```
a Warning arise and only the `*.env` file is generate in `tmp` folder. I've just installed the Acoustics Toolbox on Ubuntu 20.04.6 and tested it on Matlab, but I would try to use it also in Python 3. I tried to follow your previous instruction, but the link about your tested version of at now dosen't work. Do you have some suggestion about this issue, or someone had the same problem? Thanks in advance!

mchitre, 2023-10-18:
No other files generated in the `/tmp/` folder? If so, check if the `bellhop.exe` is in path and can be run from Python.

vinz-uts, 2023-10-18:
Hi @mchitre, thanks for your reply! I created a `lib` folder in `at` directory and I copied there all the `*.exe` files, included the `bellhop.exe` file. Then I add this folder in my path by typing `export PATH="~/...../at/lib:$PATH"`. Please, can you better explain to me how I can check if I'm able to run `bellhop.exe` from Python on Ubuntu? Thanks for you support!

vinz-uts, 2023-10-18:
Thank you @mchitre. So I already test it and it works, I have your same results: `['bellhop']`. But `pm.compute_eigenrays` function don't generate other files in `tmp` folder...

mchitre, 2023-10-18:
Then there should be at least a prt file telling you that there is an error, if you run with debug set to True

vinz-uts, 2023-10-18:
No `*prt` file is generated, only the `*.env` file was generated when I execute the code posted in my first question (that as `debug=True`). I just try to execute it now again.

mchitre, 2023-10-19:
Glad it's resolved! And thanks for posting the details of what the issue was.

jrmckenna, 2024-05-05:
@mchitre i am having the same exact problem as @vinz-uts, but trying to run on windows 10.  I've added the path variables (to point to the bellhop.exe, but [bellhop] is working from python, but only the *.env file is being created. 
My paths are here
[image].

here is how they show up;
[image]

Any idea?   Thank you so much for your time.

mchitre, 2024-05-06:
Make sure your PATH is set to point to bellhop for the environment in which you run Python. You should be able to check the environment variable from your Python REPL/notebook.