org-arl/arlpy #49: Error with pm.models()

John-Ragland, 2020-08-16:
Hi,

I've just installed Bellhop and am trying to run the notebook https://arlpy.readthedocs.io/en/latest/_static/bellhop.html to verify functionality.

I have compiled bellhop using gfortran, and have added Bellhop.exe to the PATH (trying both copying and pasting the file to the working directory and by using below sequece)
`import sys
sys.path.append(bellhop_dir)`

When I run `pm.models()`, the output is None. I'm not sure what the issue is.

Thanks for the help!

John-Ragland, 2020-08-16:
Actually, the output is an empty list. My bad.

mchitre, 2020-08-17:
Getting an empty list means that arlpy didn't find bellhop.exe. Just try running bellhop.exe manually on the command line where you run Python from, and see if it finds it.

John-Ragland, 2020-08-17:
It looks like I confused the $PATH with the python path. I added the path to bellhop.exe to $PATH and now I get the correct list: ['bellhop'].

Thanks!

mchitre, 2020-08-17:
Great! Good luck with your modeling!!

closed 2020-08-17 by mchitre