org-arl/arlpy #50: pm.plot_rays(rays, env=env, width=900) does not work with arlpy

ambrinherz, 2020-08-21:
Hello,

I was using arlpy during Sep. to Oct. last year (using python version 2x)  and it was working fine. But I see that you made an update to the library in April. I installed alrpy and Acoustic Toolbox today and I have been trying to plot rays and arrivals but it does not work. I used a simple code as follows,

**import arlpy.uwapm as pm
import arlpy.plot as plt
import numpy as np
env = pm.create_env2d()
rays = pm.compute_eigenrays(env, model='bellhop')
pm.plot_rays(rays, env=env, width=900)**

I get the below error.

**[WARN] Bellhop did not generate expected output file
Traceback (most recent call last):
  File "model.py", line 7, in <module>
    pm.plot_rays(rays, env=env, width=900)
  File "/python/Plot_test/venv/lib/python3.8/site-packages/arlpy/uwapm.py", line 478, in plot_rays
    rays = rays.sort_values('bottom_bounces', ascending=False)
AttributeError: 'NoneType' object has no attribute 'sort_values'**

I looked up online and I understand that "NoneType" means no value is passed to sort_values and bellhop did not generate an output file to be passed to pm.plot_rays method. I am not sure if your recent update in April affected the arlpy library. Could you please check and help me.

I am using python version - 3.8.5 on Mac OSX. I installed arlpy using pip (pip install arlpy). Are there any dependencies that I should add for python version - 3.8.5 ? Could you please let me know. Appreciate your help.

haozhao10, 2020-08-21:
Hi，friends.  Do you know how to install Acoustic Toolbox  of python version on Windows 10 ?

ambrinherz, 2020-08-26:
> Check if your Bellhop installation is working OK. Run it on the command
> line and see what it says. Also, if you pass in a debug=True flag to
> pm.compute_eigenrays, it'll retain the Bellhop input/output files for you
> to manually check. See if there are any errors in it.
>
> I've had a few cases of users on OS X not being able to run Bellhop
> correctly using the gfortran install for version 10+. I use gfortran
> 6.3.0 on OS X, and that works fine.

Hello Mandar,

Thank you for the prompt response. I checked again and saw that bellhop.exe
is on my PATH but when I run "bellhop.exe" from the command line it gives
me a segmentation fault. I have GFortran version 9.1.0 installed. Do you
have any idea about what causes segmentation fault? I tried adding
debug=True flag but it does not create any input/output files for me. I get
the same error as before. Please see the below screenshot. Appreciate your
help.

Plot_test python model.py

[DEBUG] Model: bellhop

[WARN] Bellhop did not generate expected output file

[DEBUG] Bellhop working files:
/var/folders/zl/x7khdv0s3ljg3bv_p8r2_y_h0000gn/T/tmp851srvsm.*

Traceback (most recent call last):

  File "model.py", line 7, in <module>

    pm.plot_rays(rays, env=env, width=900)

  File
"/Users/ambrin/Desktop/python/Plot_test/venv/lib/python3.8/site-packages/arlpy/uwapm.py",
line 478, in plot_rays

    rays = rays.sort_values('bottom_bounces', ascending=False)

AttributeError: 'NoneType' object has no attribute 'sort_values'

mchitre, 2020-08-27:
The segmentation fault seems to be the culprit. I've seen this happen with gfortran 10, and fixed it by switching to gfortran 6.3.0. Since `bellhop.exe` comes from OALIB, there isn't anything `arlpy` about it. You might want to either use an older gfortran or get in touch with Bellhop's author to see how to resolve the problem.

closed 2020-08-27 by mchitre

GT-OP, 2023-11-18:
Hello @mchitre sir actually i am bit confused about how to solve this gfortran problem can you help. Do i have to download it and add it to environment variables ? Can you help

mchitre, 2023-11-18:
Yes, I believe you need `gfortran` libraries installed on your machine to run bellhop. I am not the author of bellhop, and so can't provide you support on installing and running bellhop. `arlpy` provides a Python wrapper only, and you need a working copy of `bellhop` to use it.

P.S. You may want to look at https://github.com/org-arl/AcousticsToolbox.jl as a Julia alternative to `arlpy`, that I also maintain. That comes with pre-built binaries for bellhop for most systems, and might be easier to get started with, if you're having trouble installing bellhop from scratch.

GT-OP, 2023-11-18:
Sure sir
Thank you for the reply
Regards

> you commented.Message ID: ***@***.***>
>

GT-OP, 2023-11-20:
@mchitre sir can you share the files installed in your jupyter notebook or environment setup to run this bellhop code in jupyter notebook.

GT-OP, 2023-11-20:
@mchitre sir if it is possible. Can you upload a video on youtube on how to set up bellhop channel in python from scratch (Gfortran too) because there isn't any resources available to it. 
It will be really helpful if it could be done

mchitre, 2023-11-20:
I'm sorry @GT-OP, I cannot provide support to install `bellhop.exe`, which is NOT part of `arlpy`. You need a working install of that to use this package. You may ask the author of `bellhop` for help, if you wish to.