org-arl/arlpy #51: Range Dependent Sound Speed Profile Error

John-Ragland, 2020-09-02:
I am trying to run a simulation using a range dependent sound speed profile. I am working on my own data, but for this post I will use example code modified from arlpy documentation and the [Bellhop Example Notebook](https://arlpy.readthedocs.io/en/latest/_static/bellhop.html)

So setting the bathymetry and sound speed profile (I should note that ssp comes from arlpy docs with the last depth of profile incremented by 1 due to error I received):
```
bathy = [
    [0, 30],    # 30 m water depth at the transmitter
    [300, 20],  # 20 m water depth 300 m away
    [1000, 25]  # 25 m water depth at 1 km
]

ssp = pd.DataFrame({
          0: [1540, 1530, 1532, 1533],     # profile at 0 m range
        500: [1540, 1535, 1530, 1533],     # profile at 100 m range
        1000: [1530, 1520, 1522, 1525] },   # profile at 200 m range
        index=[0, 10, 20, 31])             # depths of the profile entries in m
```
and then creating the 2D environment:
```
env = pm.create_env2d(
    depth=bathy,
    soundspeed=ssp,
    bottom_soundspeed=1450,
    bottom_density=1200,
    bottom_absorption=1.0,
    tx_depth=15
)
```

I get the following error message when computing the eigenrays `rays = pm.compute_eigenrays(env)`:
```
None
[WARN] Bellhop did not generate expected output file
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-8-1af3b30f110e> in <module>
      1 rays = pm.compute_eigenrays(env)
----> 2 pm.plot_rays(rays, env=env, width=900)

~/opt/anaconda3/envs/ooi/lib/python3.7/site-packages/arlpy/uwapm.py in plot_rays(rays, env, invert_colors, **kwargs)
    477     >>> pm.plot_rays(rays, width=1000)
    478     """
--> 479     rays = rays.sort_values('bottom_bounces', ascending=False)
    480     max_amp = _np.max(_np.abs(rays.bottom_bounces)) if len(rays.bottom_bounces) > 0 else 0
    481     if max_amp <= 0:

AttributeError: 'NoneType' object has no attribute 'sort_values'
```

mchitre, 2020-09-03:
I suspect this is the same problem as https://github.com/org-arl/arlpy/issues/50.

Are you running on a Mac OS X with a recent version of gfortran? Try running `bellhop.exe` on the command line and see if it works properly. If you get a segmentation fault, the problem is with the compilation of Bellhop. It is solved by using an older version of gfortran on OS X. See the above referenced issue for details.

mchitre, 2020-09-03:
Actually perhaps not. I just tried this on my install of bellhop, and the problem I get is:
```
STOP Fatal Error: See print file for details
[DEBUG] Model: bellhop
At line 783 of file sspMod.f90 (unit = 5, file = '/var/folders/d5/wwnqgqpj13s503lrdb4pp0680000gn/T/tmpw8uw6yz9.env')
Fortran runtime error: Bad real number in item 1 of list input
[WARN] Bellhop did not generate expected output file
```
So bellhop does not like something in the env file. I'm checking on this.

---

For reference and debugging.

Environment file:
```
'arlpy'
25000.000000
1
'QVWT'
1 0.0 30.000000
0.000000 1540.000000 /
10.000000 1530.000000 /
20.000000 1532.000000 /
31.000000 1533.000000 /
'A*' 0.000000
30.000000 1450.000000 0.0 1.200000 1.000000 /
1
15.000000 /
1
10.000000 /
1
1.000000 /
'E'
0
-80.000000 80.000000 /
0.0 30.300000 1.010000
```

Output file (prt):
```
 BELLHOP/BELLHOP3D

 BELLHOP- arlpy
 frequency =  0.2500E+05 Hz

 Dummy parameter NMedia =            1

     Quad approximation to SSP
     Attenuation units: dB/wavelength
     THORP attenuation added
     VACUUM

 Depth =      30.00 m

 Sound speed profile:
   z (m)     alphaR (m/s)   betaR  rho (g/cm^3)  alphaI     betaI

      0.00      1540.00      0.00     1.00       0.0000    0.0000
     10.00      1530.00      0.00     1.00       0.0000    0.0000
     20.00      1532.00      0.00     1.00       0.0000    0.0000
     31.00      1533.00      0.00     1.00       0.0000    0.0000
```

mchitre, 2020-09-03:
Figure out the problem. Two things:
1.  Your last depth entry in `ssp` (31) must match the depth (30), otherwise Bellhop doesn't seem to like it.
2. For some reason, Bellhop seems to go a little beyond the receiver range while tracing, and so your `ssp` needs to be defined beyond 1000 m.

Side note: Bellhop manual recommends that the bathymetry be defined at same x ranges as the `ssp`, so you should have 500 instead of 300 in the bathymetry. This isn't a requirement to run the code, so I didn't make the change, but it is a recommendation for accuracy.

So I made the changes to your code (see comments) for both, and now it works:
```
bathy = [
    [0, 30],    
    [300, 20],               # do note that Bellhop recommends that this be at 500, same as ssp
    [1000, 25]  
]

ssp = pd.DataFrame({
          0: [1540, 1530, 1532, 1533],     
        500: [1540, 1535, 1530, 1533],     
        1000: [1530, 1520, 1522, 1525],
        1100: [1530, 1520, 1522, 1525] },       # added to go beyond 1000 m
        index=[0, 10, 20, 30])                          # changed 31 to 30

env = pm.create_env2d(
    depth=bathy,
    soundspeed=ssp,
    bottom_soundspeed=1450,
    bottom_density=1200,
    bottom_absorption=1.0,
    tx_depth=15
)

rays = pm.compute_eigenrays(env)
```

mchitre, 2020-09-03:
Closing the issue. If you still have problems, you can report details here and re-open.

closed 2020-09-03 by mchitre

m-r-tanha, 2024-01-09:
Hi Dr. Mandar
I have the same issue on the present example codes in the arlpy
I mean when I run the below code:

env = pm.create_env2d()
rays = pm.compute_eigenrays(env)
pm.plot_rays(rays, width=1000
I encounter the ([WARN] Bellhop did not generate expected output file) error with this detail
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[39], line 3
      1 env = pm.create_env2d()
      2 rays = pm.compute_eigenrays(env)
----> 3 pm.plot_rays(rays, width=1000)

 in plot_rays(rays, env, invert_colors, **kwargs)
    461 def plot_rays(rays, env=None, invert_colors=False, **kwargs):
    462     """Plots ray paths.
    463 
    464     :param rays: ray paths
   (...)
    476     >>> pm.plot_rays(rays, width=1000)
    477     """
--> 478     rays = rays.sort_values('bottom_bounces', ascending=False)
    479     max_amp = _np.max(_np.abs(rays.bottom_bounces)) if len(rays.bottom_bounces) > 0 else 0
    480     if max_amp <= 0:

AttributeError: 'NoneType' object has no attribute 'sort_values'

I have another issues like: "AttributeError: 'NoneType' object has no attribute 'time_of_arrival'"
Or: AttributeError: 'NoneType' object has no attribute 'columns'
When I run the example of arlpy in its original page

mchitre, 2024-01-29:
Try with the latest BELLHOP, and if problem persists, please post the generated input/output BELLHOP files for debugging.