org-arl/arlpy #53: ENVFile = /tmp/tmpf0e3pc5z.env   *** FATAL ERROR ***  Generated by program or subroutine: BELLHOP - READIN  Unable to open the environmental file

abhiram-gorla, 2020-09-30:
[image]
ValueError: could not convert string to float: '   0.00000000       5.00000000    \n'
I am using ubuntu18.04
python3.8
i have been trying to run bellhop model for under water channel simulation .
I have install the at_Linux.taz.gz  for acoustic toolbox
please help

mchitre, 2020-09-30:
Please run the `compute_eigenrays()` call with a `debug=True` argument. That should retain the `env` and `prt` files in the `tmp` folder for the `bellhop` call. Post those here to see what might be the problem.

abhiram-gorla, 2020-09-30:
[image]

abhiram-gorla, 2020-09-30:
[image]

abhiram-gorla, 2020-09-30:
'arlpy'
25000.000000
1
'SVWT'
1 0.0 25.000000
0.0 1500.000000 /
25.000000 1500.000000 /
'A' 0.000000
25.000000 1600.000000 0.0 1.600000 0.100000 /
1
5.000000 /
1
10.000000 /
1
1.000000 /
'E'
0
-80.000000 80.000000 /
0.0 25.250000 1.010000

.env file

 BELLHOP- arlpy                                                                  
 frequency =  0.2500E+05 Hz

 Dummy parameter NMedia =            1

     SPLINE approximation to SSP
     Attenuation units: dB/wavelength
     THORP attenuation added
     VACUUM

 Depth =    25.000000000000000      m

 Spline SSP option

 Sound speed profile:
      0.00      1500.00
     25.00      1500.00

                                 ( RMS roughness =   0.00     )
     ACOUSTO-ELASTIC half-space
     25.00      1600.00      0.00     1.60       0.1000    0.0000

 Number of sources   =            1
 Source depths (m)
   5.00000    

 Number of receivers =            1
 Receiver depths (m)
   10.0000    

 Number of ranges   =            1
 Receiver ranges (km)
   1.00000    

 Eigenray trace run
 Geometric hat beams
 Point source (cylindrical coordinates)
 Rectilinear receiver grid: Receivers at rr( : ) x rd( : )

 Number of beams   =         5000
 Beam take-off angles (degrees)
  -80.0000      -79.9680      -79.9360      -79.9040      -79.8720    
  -79.8400      -79.8080      -79.7760      -79.7439      -79.7119    
  -79.6799      -79.6479      -79.6159      -79.5839      -79.5519    
  -79.5199      -79.4879      -79.4559      -79.4239      -79.3919    
  -79.3599    

 Step length,     deltas =    2.5000000000000000      m

 Maximum ray Depth, zBox =    25.250000000000000      m
 Maximum ray range, rBox =    1010.0000000000000      m
 No beam shift in effect

 Tracing beam            1  -80.000000000000000     

 Minimum width beams                     
 halfwidth  =    4.3701937223683114     
 EpsOpt     =                (0.0000000000000000,1500000.0000000000)
 EpsMult    =    1.0000000000000000     

 Tracing beam          101  -76.799359871974374     
 Tracing beam          201  -73.598719743948791     
 Tracing beam          301  -70.398079615923166     
 Tracing beam          401  -67.197439487897583     
 Tracing beam          501  -63.996799359871964     
 Tracing beam          601  -60.796159231846360     
 Tracing beam          701  -57.595519103820770     
 Tracing beam          801  -54.394878975795159     
 Tracing beam          901  -51.194238847769554     
 Tracing beam         1001  -47.993598719743950     
 Tracing beam         1101  -44.792958591718346     
 Tracing beam         1201  -41.592318463692742     
 Tracing beam         1301  -38.391678335667130     
 Tracing beam         1401  -35.191038207641526     
 Tracing beam         1501  -31.990398079615922     
 Tracing beam         1601  -28.789757951590317     
 Tracing beam         1701  -25.589117823564713     
 Tracing beam         1801  -22.388477695539109     
 Tracing beam         1901  -19.187837567513501     
 Tracing beam         2001  -15.987197439487895     
 Tracing beam         2101  -12.786557311462291     
 Tracing beam         2201  -9.5859171834366865     
 Tracing beam         2301  -6.3852770554110823     
 Tracing beam         2401  -3.1846369273854771     
 Tracing beam         2501   1.6003200640128251E-002
 Tracing beam         2601   3.2166433286657332     
 Tracing beam         2701   6.4172834566913384     
 Tracing beam         2801   9.6179235847169444     
 Tracing beam         2901   12.818563712742547     
 Tracing beam         3001   16.019203840768153     
 Tracing beam         3101   19.219843968793761     
 Tracing beam         3201   22.420484096819365     
 Tracing beam         3301   25.621124224844969     
 Tracing beam         3401   28.821764352870574     
 Tracing beam         3501   32.022404480896178     
 Tracing beam         3601   35.223044608921782     
 Tracing beam         3701   38.423684736947386     
 Tracing beam         3801   41.624324864972998     
 Tracing beam         3901   44.824964992998602     
 Tracing beam         4001   48.025605121024206     
 Tracing beam         4101   51.226245249049811     
 Tracing beam         4201   54.426885377075415     
 Tracing beam         4301   57.627525505101012     
 Tracing beam         4401   60.828165633126616     
 Tracing beam         4501   64.028805761152228     
 Tracing beam         4601   67.229445889177839     
 Tracing beam         4701   70.430086017203436     
 Tracing beam         4801   73.630726145229048     
 Tracing beam         4901   76.831366273254645     

 CPU Time =       0.151    s

.prt file

mchitre, 2020-10-01:
Seems that the Bellhop run was successful. That should have generated a `.ray` file. Could you upload that as well please?

abhiram-gorla, 2020-10-01:
the .ray file empty

abhiram-gorla, 2020-10-01:
'BELLHOP- arlpy                                    '                                                                      25000.000000000000                                                                                                              1           1           1                                                                                            5000           1                                                                                                   0.0000000000000000                                                                                                      25.000000000000000                                                                                                    'rz'                                                                                                                                                                                                                                                                                                                                                                              

.ray file---------------------------------------------------------------------------------------------

[file.zip](https://github.com/org-arl/arlpy/files/5314357/file.zip)
zip for all three files

abhiram-gorla, 2020-10-01:
[file.zip](https://github.com/org-arl/arlpy/files/5314361/file.zip)
zip file for .ray .prt .env files
please check

mchitre, 2020-10-06:
If you look at the `prt` file, you'll find the reason for the failure:
```
 *** WARNING ***
 Generated by program or subroutine: TraceRay2D
 Insufficient storage for ray trajectory
```
It seems that Bellhop is unable to store all the ray trace for the number of beams you have specified.

patel999jay, 2020-10-22:
@abhiram-gorla : Strange, I ran your files, and perfectly run on my machine. Attached are the results. 

[test-arlpy.zip](https://github.com/org-arl/arlpy/files/5424720/test-arlpy.zip)

closed 2020-11-05 by mchitre