org-arl/arlpy #54: `arrival_number` backwards

Erhannis, 2020-11-04:
In https://arlpy.readthedocs.io/en/latest/_static/bellhop.html , it indicates ("...the first 10 arrivals...") that `arrival_number` gives the order in which the rays arrived.  However, the associated code (the one with `arrivals[arrivals.arrival_number < 10]`) shows a resulting table with the _last_ 10 arrivals - arrival 0, for instance, has many bounces and a `time_of_arrival` of 0.721796, which decrease as `arrival_number` increases.

mchitre, 2020-11-05:
The arrivals are returned in the order they are generated by Bellhop in its output file. `arlpy` does not do any sorting on the loaded data. You can easily sort the pandas frame that `compute_arrivals()` returns by time, if you need that. The `arrival_number` is simply an index in the Bellhop's output file, and not meant to be used as a timing index.

The phrase "first 10 arrivals" could be misleading, since I meant "first 10" in the arrival list, not by chronology. I have updated the documentation to accurately reflect this: "... for the first 10 entries in the arrivals list (not necessarily sorted by time)".

closed 2020-11-05 by mchitre

Erhannis, 2020-11-05:
Oh, ok - thanks!