org-arl/arlpy #55: arrivals_to_impulse_response incorrect time scale

Erhannis, 2020-11-04:
As shown in In https://arlpy.readthedocs.io/en/latest/_static/bellhop.html , it seems like arrivals_to_impulse_response gives an incorrect time scale.  According to the preceding table and graph, the final arrival is recorded at 0.721796 seconds, but the plot of `arrivals_to_impulse_response` shows the final arrival at ~0.055 seconds.  Likewise, the output of `arrivals_to_impulse_response` (`ir`) only contains 5293 elements, but given a total time of 0.721796 seconds and a rate of 96000 Hz, I'd have instead expected ~69292 elements.

Erhannis, 2020-11-05:
Oh, no wait; I was wrong - I didn't notice that the first graph X _starts_ at around 0.665, making the whole thing ~0.055 seconds wide.  `arrivals_to_impulse_response` just shifts everything over such that the first arrival starts at t = 0, I think; sorry.

closed 2020-11-05 by Erhannis

mchitre, 2020-11-05:
@Erhannis `arrivals_to_impulse_response()` takes a keyword argument `abs_time`, which defaults to `False`. If you pass `True`, it'll give you in absolute time, according to the expectations you outlined in your original message.

Erhannis, 2020-11-05:
Oh, neat; thanks