org-arl/arlpy #56: Surface reflection coefficient

GogaGeosal, 2020-11-04:
I am trying to model long-range sound propagation under the surface sea-ice. It appears from the bellhop manual that it is possible to provide a surface reflection coefficient, but this does not seem to be supported by arlpy yet?

mchitre, 2020-11-05:
I'll add it in as part of the next release, or if you need it earlier and decide to patch the code, I'll be happy to accept a pull request.

GogaGeosal, 2020-11-09:
Hi Mandar -- many thanks for the response. I will patch the code and will request a pull so that I can proceed with my coding. Many thanks again.

In the meantime, I cannot get my head around why I am getting an annoying FATAL error for a relatively simple environment- would appreciate your help.

I have got the following simple script:

    bathy = np.array([100, 200, 250, 220, 150, 400])
    depth = np.array([r for r in zip(np.linspace(0, 1000, len(bathy)), bathy)])
    ss_profile = 1500 * np.ones((10, 8))
    depth_level = np.array([0, 50, 100, 150, 200, 250, 300, 350, 380, 400])
    #ssp = np.vstack([depth_level, np.mean(ss_profile, axis=1)]).T
    ssp = pd.DataFrame(ss_profile, index=depth_level, columns=np.linspace(0, 1100, len(ss_profile[0])))
    env = pm.create_env2d(depth_interp='curvilinear',
                          depth=depth,
                          soundspeed=ssp,
                          tx_depth=20,
                          rx_depth=25,
                          rx_range=1000
                          )
    rays = pm.compute_eigenrays(env)
    pm.plot_rays(rays, env=env, width=900)

where I have got some bathy data and some depth and range dependent sound speed (in fact it is always 1500 m/s, but tis does not really matter). If I run the code as it is, it works and it generates a plot. 

If i instead I make the environment deeper at the begining of the range-axis, for example:
 bathy = np.array([250, 200, 250, 220, 150, 400])
it gives a fatal error. Also, using the same bathy, if you comment the data frame and uncomment the range-indepedent ssp, the fatal error disappears. Any idea why the change in the bathy causes the range-dependent sound speed to cause fatal errors?

Thanks in advance!

mchitre, 2020-11-10:
The error message suggests "ray is outside the box where the soundspeed is defined", and I suspect this is exactly the problem. Bellhop is quite stringent requirements on depth in soundspeed profile and bathymetry being consistent.

GogaGeosal, 2020-11-10:
Thanks. This is true that the model has some strict requirements. 

I think I found the problem, which is quite weird since it occurs only for range-depdent sound speed profile. Effectively, one need to give a negative starting range (rather than starting from zero). For the example above, the following parametrization works (just in case someone else has the same issue): 
ssp = pd.DataFrame(ss_profile, index=depth_level, columns=np.linspace(-400, 1100, len(ss_profile[0]))).

Of course, this implies that one will need to artificially correct the induced offset in the sound speed profile.

Thanks.

frankMcQuarrie, 2023-09-09:
@GogaGeosal ! Thank you for asking this question. This is a few years after but I'm encountering the same issue; I have a range dependent sound profiles and I'm getting the "ray is outside the box where soundspeed is defined" error.

I'm not sure you still check this but wanted to ask, what values did you make negative? I use the .exe file so I'm feeding in .env, .ssp, and .bty files. For the sound speed profile I tried making my first value negative but no luck.