org-arl/arlpy #58: one question about _select_model

haozhao10, 2020-11-23:
File "../uwapm.py", line 570, in _select_model
    raise ValueError('No suitable propagation model available')
ValueError: No suitable propagation model available

mchitre, 2020-11-23:
You need bellhop installed on your system for it to be available in arlpy

haozhao10, 2020-11-25:
Thanks a lot. I have fixed this problem.

closed 2020-11-25 by mchitre

gu5b4g, 2023-05-27:
> Thanks a lot. I have fixed this problem.

Hey, I know this is an old thread - so a long shot; but how did you go about fixing this problem? 
I used the GFortran compiler on my Mac - and am getting the same issue when I follow along on https://arlpy.readthedocs.io/en/latest/_static/bellhop.html
I am thinking it is to do with Bellhop being an exe and me attempting to run it on a Mac; but just incase its a simple fix before I go using parallels, I'd really appreciate any hints.
Thanks in advance!

mchitre, 2023-05-27:
I do run Bellhop on a Mac. To do so, you need to install GFortran with the right versions of the libraries and then compile Bellhop.