org-arl/arlpy #62: Error when saving .png from bokeh interface

John-Ragland, 2021-03-15:
When I select the save button a ray plot (below)
[image]

I get the following .png, which is the top left of the image that I want.
[image]

Additionally is there a way that I can save an image in a custom file format (such as .eps)?

mchitre, 2021-03-15:
I am unable to reproduce your problem, as the file saves fine for me. I'm guessing this is some incompatibility with the Bokeh library and your setup. Could you try using Bokeh directly for plots and see if those save fine?

I personally tend to use a screen capture to save the plot, rather than the Bokeh save option. That allows me to save in any format I want too.

John-Ragland, 2021-03-15:
I get the same error when just using Bokeh, so it looks like this is a bug with Bokeh.

As for saving a plot, I would like to programmatically create figures for a publication that are vectorized. Is there any support for programmatically saving the figures? (such as the equivalent to matplotlib's matplotlib.pyplot.savefig())?

John-Ragland, 2021-03-15:
I'm not sure how I could use the bokeh support for vectorized images within arlpy.

Thanks for the help!

mchitre, 2021-03-15:
Yes, you can programmatically save the figure. See Bokeh link I posted above.

John-Ragland, 2021-04-15:
I'm unsure how to get arlpy.plot.gcf() to work. This is what I'm trying right now
after creating an environment variable (env):

```
import arlpy.uwapm as pm

rays = pm.compute_eigenrays(env)
pm.plot_rays(rays, env=env, width=700)
fig = arlpy.plot.gcf()
```

arlpy.plot.gcf() returns None

mchitre, 2021-04-16:
You'll need to hold the plot to get the handle.

Try this:
```
import arlpy.uwapm as pm
import arlpy.plot as plt

rays = pm.compute_eigenrays(pm.create_env2d())

plt.hold(True)
pm.plot_rays(rays, env=pm.create_env2d(), width=700)
fig = plt.gcf()
plt.hold(False)
```

John-Ragland, 2021-04-16:
This fixed it for me thanks!

closed 2021-04-16 by John-Ragland