org-arl/arlpy #65: Plotting issues

kerentalmon, 2021-05-20:
a. Running great on jupyter notebook but not plotting on other IDE e.g. Spyder
b. Plots saving format - cannot save into PDF

mchitre, 2021-05-20:
Plots in arlpy are made using [Bokeh](https://docs.bokeh.org/en/latest/index.html), so you might want to check whether your Spyder install works with Bokeh.

kerentalmon, 2021-05-23:
Bokeh displays on IPython. To use on IDE such as Spyder, the plotting commands has to be rerun on the interactive console and then web-browser opens to display the plots

mchitre, 2021-05-24:
So doing that resolves the problem for you?

kerentalmon, 2021-05-24:
The above solution is kind of a makeshift and may not be complete. For my needs I updated arlpy plot.py as follows:
in 
`class figure:`
under 
`def __enter__(self):`
I added the following
```
         _bplt.output_file(filename="custom_filename.html")  # temp name for the figure
        p = _figure  
        _bplt.save(p)  # save the figure
        webbrowser.open('custom_filename.html')  # display in browser
```
at the top I added
`import webbrowser`

I did not check for conflicts with jupyter notebook. The added code saves any figure as temporary file (using bokeh) and later shows it in the default computer browser. I did not implement it in `class many_figures:` and cannot ascertain there isn't other issues that might arise.

mchitre, 2021-05-24:
Seems like a reasonable workaround for your needs, and we'll leave it here in case someone else needs it.

I believe this is currently being addressed by Spyder team in https://github.com/spyder-ide/spyder/issues/8367, and when that issue is closed, Spyder should work with Bokeh plots (and therefore automatically with arlpy plots). So no further action required here.

closed 2021-05-24 by mchitre