org-arl/arlpy #68: Question about bellhop install in linux

andy88980, 2021-06-19:
Hi,
I dowloaded the at.zip for Linux and makefile.
And the I used this command( export PATH=$PATH:/mypath/at_2020_11_4/Bellhop )to add the PATH.
But I still got the following error:
ValueError No suitable propagation model available

How to fix this probelm?

Thank you.
Best regard.

mchitre, 2021-06-19:
Does typing `bellhop.exe` on command line run Bellhop? If so, make sure you start Python or Jupyter from the same terminal where this works.

P.S. You may need to put this in your `.bashrc` or `.profile` file to ensure the `PATH` is set correctly every time you start a terminal.

andy88980, 2021-06-20:
Thank you, I think it work.
But when I trying to run an example file at_2020_11_4/tests/arctic/arctic.env, it came up with 'Stop Fatal Error:Check the print file for detail.'
Here is the content of env file and print file.

arctic.env
'Arctic profile'	! TITLE
500.0 			! frequency (Hz)
1
'SVF'
3750  0.0  3750.0
   0. 1436.0     0.0  1.0   /
 100. 1441.9     0.0  1.0   /
 150. 1450.0     0.0  1.0   /
 200. 1458.4     0.0  1.0   /
 250. 1460.0     0.0  1.0   /
 300. 1460.5     0.0  1.0   /
 400. 1461.0     0.0  1.0   /
 450. 1461.5     0.0  1.0   /
 500. 1462.0     0.0  1.0   /
1000. 1466.7     0.0  1.0   /
2000. 1479.6     0.0  1.0   /
2500. 1487.9     0.0  1.0   /
3750. 1510.4     0.0  1.0   /
'A'  0.0
  3750.  1510.4     0.0  1.0 /  
0.0  1504.0
300.0			! RMAX (km)
1 
100.0 /		! NSRCS  SD(1:NSRCS)
1 
200.0 /		! NRCVRS  RD(1:NRCVRS)
------------------------------------------------------------------------------
arctic.env.prt
 BELLHOP/BELLHOP3D

 ENVFile = /mypath/at_2020_11_4/Bellhop/arctic.env.env

 *** FATAL ERROR ***
 Generated by program or subroutine: BELLHOP - READIN
 Unable to open the environmental file

andy88980, 2021-06-20:
Also, I try to run a python code with arlpy.
Like this:

import arlpy.uwapm as pm
import arlpy.plot as plt

bathy = [
        [0, 30],  # 30 m water depth at the transmitter
        [300, 20],  # 20 m water depth 300 m away
        [1000, 25]  # 25 m water depth at 1 km
    ]
    ssp = [
        [0, 1540],  # 1540 m/s at the surface
        [10, 1530],  # 1530 m/s at 10 m depth
        [20, 1532],  # 1532 m/s at 20 m depth
        [25, 1533],  # 1533 m/s at 25 m depth
        [30, 1535]  # 1535 m/s at the seabed
    ]
    env = pm.create_env2d(depth=bathy,
                          soundspeed=ssp,
                          bottom_soundspeed=1450,
                          bottom_density=1200,
                          bottom_absorption=1.0,
                          tx_depth=15)
    pm.print_env(env)
    rays = pm.compute_rays(env)
    pm.plot_rays(rays, width=1000)

It also came up with:'Stop Fatal Error:Check the print file for detail.'

How can I fix this?
Thank you for your help.

mchitre, 2021-06-20:
For your problem re running Bellhop directly, the clue is in the error output: `Unable to open the environmental file` and `/mypath/at_2020_11_4/Bellhop/arctic.env.env`. Note the `.env.env` that Bellhop is trying to read, whereas your env file probably has only one `.env`. Bellhop automatically assumes the `.env` and so when running Bellhop, you should run it as `bellhop.exe /mypath/at_2020_11_4/Bellhop/arctic` without the `.env`.

For any further guidance on using Bellhop without arlpy, please consult Bellhop's documentation, and if you have doubts, contact Bellhop's author.

mchitre, 2021-06-20:
For your problem running Bellhop from arlpy, try a simpler environment example (e.g. from the docs) to check if your installation is fine. If it is, you can use `rays = pm.compute_rays(env, debug=True)` to ask arlpy to retain the Bellhop print file for you to check what the error might be.

andy88980, 2021-06-21:
Ok, Thank you for your answer.
I think think the problem solved.

mchitre, 2021-06-21:
Great!

closed 2021-06-21 by mchitre