org-arl/arlpy #70: Not an issue just a helping hand request

StuartIanNaylor, 2021-07-01:
I am struggling to get my head round the required input numpy shape and the pos shape for a 2 mic linear (96mm spacing)
Ignore the below pos as was one of many.
Is it because the rec.shape (3200, 2) is wrong?

Apols all but any help would be appreciated.

```
from arlpy import bf
import numpy as np
import sounddevice as sd

# narrowband (1 kHz) timeseries array data assumed to be loaded in x
# sensor positions assumed to be in pos

fs = 16000
rec_duration = 0.20
duration = 10
sample_rate = 16000
num_channels = 2

def sd_callback(rec, frames, time, status):
  pos = [0.0, 0.0, 0.0, 0.0, 0.096, 0.0]
  y = bf.bartlett(rec, 1000, bf.steering_plane_wave(pos, 343, np.linspace(-np.pi/2, np.pi/2, 181)))
  print(y)

# Start streaming from microphone
with sd.InputStream(channels=num_channels,
                    samplerate=fs,
                    blocksize=int(fs * rec_duration),
                    callback=sd_callback):
    sd.sleep(int(duration * 1000))
```

mchitre, 2021-07-06:
What's the error you're seeing?

I'm not sure I understand the `pos`. Why are 5 sensors at the same location 0.0? I'll take your advice and ignore the `pos`, but perhaps you didn't mean to have 5 sensors? Your recording is 2-channel, based on your `rec.shape`, but your `pos` suggests 5 sensors.

StuartIanNaylor, 2021-07-08:
```
def sd_callback(rec, frames, time, status):
  print(rec.shape)
  rec = rec.reshape(2, 3200)

  pos = [-0.05, 0.05]
  y = bf.bartlett(rec, 1000, bf.steering_plane_wave(pos, 343, np.linspace(-np.pi/2, np.pi/2, 181)))
  print(y)
```
  
I am not sure why as should of realised as soundevice seems to be backwards with [audio, channels] so added a reshape and hey presto.
Apols as before it was just a quick copy+paste after trying various options.

closed 2021-07-08 by StuartIanNaylor