org-arl/arlpy #71: Range dependent SSP error

CT-Hung, 2021-09-03:
Hi authors

I use arlpy toolbox to compute the arrivals.
The toolbox and Bellhop are doing well when I input the single-point SSP.
But if I change the single-point SSP to range-dependent SSP, the following error will show up.
"[WARN] Bellhop did not generate expected output file"
I also tried the debug function and there are only .prt, .env, .bty and .ssp files.
Is there anything that I was wrong with?
If you need more information please let me know.
Thank you in advance. 

Ching Tang

mchitre, 2021-09-05:
Check what the prt file says.

CT-Hung, 2021-09-05:
Here is the prt file information:

 BELLHOP/BELLHOP3D

 BELLHOP- arlpy                                                                  
 frequency =   8800.     Hz

 Dummy parameter NMedia =            1

     Quad approximation to SSP
     Attenuation units: dB/wavelength
     THORP volume attenuation added
     VACUUM

 Depth =    3047.97 m

 Sound speed profile:
   z (m)     alphaR (m/s)   betaR  rho (g/cm^3)  alphaI     betaI

      0.00      1541.30      0.00     1.00       0.0000    0.0000
      2.00      1541.25      0.00     1.00       0.0000    0.0000
      4.00      1541.20      0.00     1.00       0.0000    0.0000
      6.00      1541.16      0.00     1.00       0.0000    0.0000
      8.00      1541.10      0.00     1.00       0.0000    0.0000
     10.00      1541.04      0.00     1.00       0.0000    0.0000
     15.00      1540.85      0.00     1.00       0.0000    0.0000
     20.00      1540.58      0.00     1.00       0.0000    0.0000
     25.00      1540.21      0.00     1.00       0.0000    0.0000
     30.00      1539.76      0.00     1.00       0.0000    0.0000
     35.00      1539.29      0.00     1.00       0.0000    0.0000
     40.00      1538.79      0.00     1.00       0.0000    0.0000
     45.00      1538.27      0.00     1.00       0.0000    0.0000
     50.00      1537.73      0.00     1.00       0.0000    0.0000
     55.00      1537.13      0.00     1.00       0.0000    0.0000
     60.00      1536.51      0.00     1.00       0.0000    0.0000
     65.00      1535.88      0.00     1.00       0.0000    0.0000
     70.00      1535.23      0.00     1.00       0.0000    0.0000
     75.00      1534.60      0.00     1.00       0.0000    0.0000
     80.00      1533.93      0.00     1.00       0.0000    0.0000
     85.00      1533.24      0.00     1.00       0.0000    0.0000
     90.00      1532.52      0.00     1.00       0.0000    0.0000
     95.00      1531.78      0.00     1.00       0.0000    0.0000
    100.00      1531.08      0.00     1.00       0.0000    0.0000
    110.00      1529.83      0.00     1.00       0.0000    0.0000
    120.00      1528.66      0.00     1.00       0.0000    0.0000
    130.00      1527.50      0.00     1.00       0.0000    0.0000
    140.00      1526.26      0.00     1.00       0.0000    0.0000
    150.00      1525.09      0.00     1.00       0.0000    0.0000
    160.00      1523.94      0.00     1.00       0.0000    0.0000
    170.00      1522.88      0.00     1.00       0.0000    0.0000
    180.00      1521.83      0.00     1.00       0.0000    0.0000
    190.00      1520.74      0.00     1.00       0.0000    0.0000
    200.00      1519.69      0.00     1.00       0.0000    0.0000
    220.00      1517.72      0.00     1.00       0.0000    0.0000
    240.00      1515.85      0.00     1.00       0.0000    0.0000
    260.00      1514.04      0.00     1.00       0.0000    0.0000
    280.00      1512.30      0.00     1.00       0.0000    0.0000
    300.00      1510.62      0.00     1.00       0.0000    0.0000
    350.00      1506.28      0.00     1.00       0.0000    0.0000
    400.00      1501.96      0.00     1.00       0.0000    0.0000
    500.00      1493.87      0.00     1.00       0.0000    0.0000
    600.00      1489.04      0.00     1.00       0.0000    0.0000
    700.00      1486.60      0.00     1.00       0.0000    0.0000
    800.00      1484.99      0.00     1.00       0.0000    0.0000
    900.00      1484.19      0.00     1.00       0.0000    0.0000
   1000.00      1483.65      0.00     1.00       0.0000    0.0000
   1100.00      1483.28      0.00     1.00       0.0000    0.0000
   1200.00      1483.15      0.00     1.00       0.0000    0.0000
   1300.00      1483.71      0.00     1.00       0.0000    0.0000
   1400.00      1484.52      0.00     1.00       0.0000    0.0000
   1500.00      1485.45      0.00     1.00       0.0000    0.0000
   1600.00      1486.51      0.00     1.00       0.0000    0.0000
   1800.00      1488.76      0.00     1.00       0.0000    0.0000
   2000.00      1491.12      0.00     1.00       0.0000    0.0000
   2200.00      1493.79      0.00     1.00       0.0000    0.0000
   2400.00      1496.70      0.00     1.00       0.0000    0.0000
   2600.00      1499.74      0.00     1.00       0.0000    0.0000
   2800.00      1502.92      0.00     1.00       0.0000    0.0000
   3000.00      1506.24      0.00     1.00       0.0000    0.0000
   3200.00      1509.66      0.00     1.00       0.0000    0.0000
   3400.00      1513.11      0.00     1.00       0.0000    0.0000

mchitre, 2021-09-05:
Is that the end of the file? If so, Bellhop run does not seem to have successfully completed.

CT-Hung, 2021-09-05:
Yes, it is the end of the file.
I have the same idea that Bellhop didn't run successfully.
Can I do anything to debug or check the problem?

mchitre, 2021-09-05:
The problem is likely in Bellhop itself, so arlpy can't _solve_ the problem for you. But I can give you some suggestions to debug...

1. I think Bellhop needs your SSP to end at the max depth. Your environment seems to have a max depth of 3047.97 m, but your SSP goes to 3400.0 m. Try changing the last point to end at 3047.97 m.
2. If that doesn't work, copy out the `.env`, `.bty` and `.ssp` files and try running Bellhop on the command line with those. Perhaps Bellhop might be crashing, but might give you some clues as to the problem on the terminal.
3. Try reducing the number of points in your SSP, as it seems to be quite dense. You can also try changing the interpolation.

CT-Hung, 2021-09-05:
Thanks for your suggestion.
1. and 3. It is still not working. The single point SSP has the same max depth of 3400.0m and the same dense distance, and it was working.
2. Bellhop was crashing, as you said. It printed the following error message, but I do not understand at all:
    PS C:\Users\cthung\Desktop\bellhopTest> .\bellhop.exe .\tmpm64tev_o
    At line 905 of file sspMod.f90 (unit = 5, file = '.\tmpm64tev_o.env')
    Fortran runtime error: Bad real number in item 1 of list input

    Error termination. Backtrace:

    Could not print backtrace: libbacktrace could not find executable to open
    #0  0xffffffff
    #1  0xffffffff
    #2  0xffffffff
    #3  0xffffffff
    #4  0xffffffff
    #5  0xffffffff
    #6  0xffffffff
    #7  0xffffffff
    #8  0xffffffff
    #9  0xffffffff
    #10  0xffffffff
    #11  0xffffffff
    #12  0xffffffff
    #13  0xffffffff
    #14  0xffffffff
    #15  0xffffffff

mchitre, 2021-09-05:
1. I don't understand what you mean by a single point SSP having the same density and max depth. If it's a single point SSP, I'm guessing it is an iso-velocity profile with just a single value for SSP. Such an entry wouldn't have density or max depth.

2. Re error:
```
At line 905 of file sspMod.f90 (unit = 5, file = '.\tmpm64tev_o.env')
    Fortran runtime error: Bad real number in item 1 of list input
```
suggests that Bellhop doesn't like some number in your SSP. Item 1, I believe is depth entry in the SSP. So it doesn't like a depth entry. I suspect it is that the depth entry is out of bounds (0 and max depth).

CT-Hung, 2021-09-05:
1. Sorry for my poor English.
The single point SSP means range-independent SSP, and the start depth and the max depth are 0 m and 3400 m.
The attached file is a zip file include the range-dependent.env and range-Independent.env.
[envFile.zip](https://github.com/org-arl/arlpy/files/7111150/envFile.zip)

2. Still trying

mchitre, 2021-09-05:
I believe your issue is the same as https://github.com/org-arl/arlpy/issues/51, so see if that helps.

CT-Hung, 2021-09-05:
It is helpful and the problem is solved.
Here are my comments:
1. The situation of range-dependent SSP:
The max depth of SSP needs to be totally the same as the bathy and the range of SSP needs to be a little bit larger than the bathy.
2. The situation of range-independent SSP:
The max depth of SSP can be larger than the bathy.

Thank you very much and the toolbox does help me a lot.

mchitre, 2021-09-05:
Excellent, glad it's solved!

closed 2021-09-05 by mchitre