org-arl/arlpy #73: Bellhop throws an unexpected error with arlpy default env file

patel999jay, 2022-04-19:
I am trying to trace an error that `bellhop` is throwing when it takes `arlpy default env file`. 

I don't think it is running `bellhop` incorrectly because the print file with and without `arlpy` are exactly same. 

Here is what i am using right now to test this: 
```bash
jay@jay-VirtualBox:/tmp$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04 (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
jay@jay-VirtualBox:/tmp$ python3 --version
Python 3.10.4
```
When I called the `pm.models()` from `python` or in `ipython` terminal.
bellhop returns with the error that `STOP Fatal Error: Check the print file for details`

```python
jay@jay-VirtualBox:~$ ipython3
Python 3.10.4 (main, Apr  2 2022, 09:04:19) [GCC 11.2.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.2.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: 
   ...: import arlpy.uwapm as pm
   ...: import arlpy.plot as plt
   ...: import numpy as np
   ...: import pandas as pd

In [2]: 
   ...: 
   ...: pm.models()

STOP Fatal Error: Check the print file for details
Out[2]: ['bellhop']

In [3]: env = pm.create_env2d()

In [4]: pm.print_env(env)
                name : arlpy
   bottom_absorption : 0.1
      bottom_density : 1600
    bottom_roughness : 0
   bottom_soundspeed : 1600
               depth : 25
        depth_interp : linear
           frequency : 25000
           max_angle : 80
           min_angle : -80
              nbeams : 0
            rx_depth : 10
            rx_range : 1000
          soundspeed : 1500
   soundspeed_interp : spline
             surface : None
      surface_interp : linear
            tx_depth : 5
   tx_directionality : None
                type : 2D

In [5]: pm.plot_env(env, width=900)

In [6]: rays = pm.compute_eigenrays(env)
STOP Fatal Error: Check the print file for details

In [7]: rays = pm.compute_eigenrays(env,debug=True)
    ...: pm.plot_rays(rays, env=env, width=900)
STOP Fatal Error: Check the print file for details
[DEBUG] Model: bellhop
[DEBUG] Bellhop working files: /tmp/tmpvpslno76.*

In [8]: arrivals = pm.compute_arrivals(env)
STOP Fatal Error: Check the print file for details

In [9]: pm.plot_arrivals(arrivals)

```
I checked the print files at temporary location and it looks fine. Only difference would be(that is acceptable) :

```bash
jay@jay-hp-laptop:~/Documents/stack/g$ git diff arlpy_print.prt without_arlpy_tmpvpslno76.prt
diff --git a/arlpy_print.prt b/arlpy_print.prt
index f3def08..289a310 100644
--- a/arlpy_print.prt
+++ b/arlpy_print.prt
@@ -125,4 +125,4 @@ Tracing beam    4701     70.43
 Tracing beam    4801     73.63
 Tracing beam    4901     76.83
 
- CPU Time =       0.234    s
+ CPU Time =       0.181    s
jay@jay-hp-laptop:~/Documents/stack/g$ 
```

I am attaching both files as zip if that helps.
[arlpy_run_tmpvpslno76.zip](https://github.com/org-arl/arlpy/files/8515844/arlpy_run_tmpvpslno76.zip)
[without_arlpy_tmpvpslno76.zip](https://github.com/org-arl/arlpy/files/8515845/without_arlpy_tmpvpslno76.zip)

so i thought it may be `bellhop/acoustic toolbox` throwing an error but it is running fine just as normal.

```bash
jay@jay-VirtualBox:~/Desktop$ ls
MunkS.env  MunkS.env.prt  MunkS.prt  Test.env
jay@jay-VirtualBox:~/Desktop$ bellhop.exe Test
jay@jay-VirtualBox:~/Desktop$
```

It looks like it took an empty env file first somehow and then it retrieve a temporary default *.env file or am i missing something ? 

P.S. I thought there is something wrong my old bellhop installation so i install a fresh new ubuntu 22.04 and acoustic toolbox and tested this ; also tested this with `windows 10` and `windows 11`. I am experiencing the issue on all platforms. Is there a workaround this ?

mchitre, 2022-04-20:
Nothing has changed in `arlpy` recently, so I suspect this is a problem with version mismatch between the tested `bellhop.exe` and what you're using. What version of OALIB Acoustics Toolbox have you installed?

patel999jay, 2022-04-20:
First I thought it may be something to do with my installation but I installed a fresh ubuntu and downloaded the newest version of acoustic toolbox(i guess the one updated on 2020_11_4). 

I think arlpy is running an empty env file when we call an function pm.models() and that is why it is throwing that error but works fine when we're supplying an env. I will take a closer loop tomorrow and keep updating the issue. 

Thanks for your time and help.

mchitre, 2022-04-20:
I recall there was a known issue (and perhaps a resolution) for this, but don't offhand remember details. Let me check and get back on this.

mchitre, 2022-04-29:
https://github.com/org-arl/arlpy/issues/9

mchitre, 2022-12-12:
@patel999jay has this been resolved for you?

patel999jay, 2022-12-12:
No, This has not resolved yet. I still has the same issue.

closed 2023-10-24 by patel999jay

scjuber, 2025-05-30:
Hey! I am still having this issue both in the command line and arlpy. I used the newest edition of AT from your github @patel999jay. I have Windows 11 OS but I'm running with Ubuntu 24.04.1 LTS on WSL. I get the same error building at and running from Powershell. Any advice?

patel999jay, 2025-05-31:
> I am still having this issue both in the command line and arlpy. I used the newest edition of AT from your github @patel999jay. I have Windows 11 OS but I'm running with Ubuntu 24.04.1 LTS on WSL. I get the same error building at and running from Powershell. Any advice?

Why are you running this in WSL ? You can natively run this on windows machine, just curious. Are you running the commandline from same folder as 'bellhop.exe' or did you put that in your windows path ? Or are you running this in WSL ?

scjuber, 2025-06-01:
Thanks for the response! I also tried building it for Windows 11 and also had similar issues so I thought using the Linux version might work better. I'm running everything in WSL.

patel999jay, 2025-06-01:
> Thanks for the response! I also tried building it for Windows 11 and also had similar issues so I thought using the Linux version might work better. I'm running everything in WSL.

Did you download windows binaries from this link : http://oalib.hlsresearch.com/AcousticsToolbox/atWin10_2020_11_4.zip ?

if yes then try this :

1. Option 1: Run ARLpy natively on Windows

    - Install Python & ARLpy directly on Windows.
    - Make sure `bellhop.exe` is in the PATH or the same folder from which they run Python.
    - Run everything from a native Windows terminal (e.g., PowerShell or Command Prompt).

then if you run below commands :

```python
In [1]: 
   ...: import arlpy.uwapm as pm
   ...: import arlpy.plot as plt
   ...: import numpy as np
   ...: import pandas as pd

In [2]: 
   ...: 
   ...: pm.models()
Out[2]: ['bellhop']
```

if it shows this last line `Out[2]: ['bellhop']`, that means it is configured correctly.

scjuber, 2025-06-01:
Thank you, yes, that's what I did. After pm.models() it prints 'bellhop'
and creating and plotting the 2d environment works, but when I compute rays
I get 'Bellhop did not produce the expected output'.

scjuber, 2025-06-04:
> <https://urldefense.com/v3/__https://github.com/org-arl/arlpy/issues/73*issuecomment-2931360818__;Iw!!K-Hz7m0Vt54!nwSbk6crfT67VNGsedzE1veSAV23IFTy25wZbjhOwIAedv1IW8N4YvTCZvWYn4wihkNhP0r6W11ArbTzK5Y9eA$>
>
> I was able to get it working using the precompiled binaries from
> bellhopcuda
> https://github.com/A-New-BellHope/bellhopcuda
> <https://urldefense.com/v3/__https://github.com/A-New-BellHope/bellhopcuda__;!!K-Hz7m0Vt54!nwSbk6crfT67VNGsedzE1veSAV23IFTy25wZbjhOwIAedv1IW8N4YvTCZvWYn4wihkNhP0r6W11ArbS5N9HIHQ$>.
> Could not properly read
> the environmental files any other way but it's working through arlpy now.
> Thanks for your help!
> … <#m_-2354098074765663879_>
>
> you can also try to run it in debug mode and check the print file it
> generated and it will provide you more info, i.e
>
> In [7]: rays = pm.compute_eigenrays(env,debug=True)
>     ...: pm.plot_rays(rays, env=env, width=900)STOP Fatal Error: Check the print file for details
> [DEBUG] Model: bellhop
> [DEBUG] Bellhop working files: /tmp/tmpvpslno76.*

Ok, thank you. I will try that.