org-arl/arlpy #74: How can I make sure that BELLHOP installed on my PC?

catauggie, 2022-12-23:
Hi, I am user of windows 10 OS. 

I have installed BELLHOP from Acoustics Toolbox
http://oalib.hlsresearch.com/AcousticsToolbox/
But I don't understand how can I make sure that BELLHOP installed on my PC, because when I compile code in Jupyter I have an error No suitable propagation available

mchitre, 2022-12-23:
You need to make sure that the `bellhop.exe` is in your `PATH`, so that you can run it on the command line before you run Jupyter.

catauggie, 2022-12-23:
Can I clarify: what is the PATH?

catauggie, 2022-12-23:
And what should I type in command line before run jupiter?

mchitre, 2022-12-23:
https://www.maketecheasier.com/what-is-the-windows-path/

catauggie, 2022-12-23:
so, I decided to install it again.
As I understood,
1)  I should download atWin10_2020_11_4.zip Binaries (zip file) for Windows 10 from http://oalib.hlsresearch.com/AcousticsToolbox/, right?
2) I found bellhop.exe (my folder \acoustic toolbox\atWin10_2020_11_4\atWin10_2020_11_4\windows-bin-20201102)
3) I have created PATH variable with mentioned above path
and finally the same error((

catauggie, 2022-12-23:
I  don't understand how PATH can help to avoid this error. Maybe there are some extra options to fix?

mchitre, 2022-12-23:
The `PATH` tells your computer where to find the executable for a command you're trying to run.

If you open a command prompt and type `bellhop.exe`, what do you see as output? If your `PATH` is correctly setup, it should be able to find `bellhop.exe` and run it and show you the output. If not, it'll tell you that it was not found.

If you're successfully able to run it, then running Jupyter from the same command prompt should allow `arlpy` to find `bellhop.exe` too. The message that no model is found indicates the the `PATH` for the command prompt that runs Jupyter does not have the OALIB installation in it.

catauggie, 2022-12-23:
So, If I type PATH in command line, I have:
[image]

if I type bellhop.exe, I have: "bellhop.exe" is not internal or external
command, executable program, or batch file.
[image]

catauggie, 2022-12-23:
maybe after adding bellhop.exe to PATH exist any extra actions to do?

mchitre, 2022-12-23:
The text `bellhop.exe` should be removed from the PATH. The PATH is supposed to contain the directories, not the files.

catauggie, 2022-12-23:
Yes, exactly this helped

closed 2022-12-23 by catauggie