org-arl/arlpy #75: Range dependent profile -I am getting Print('[WARN] Bellhop did not generate expected output file')

heeradshakeera, 2023-01-02:
Hai, I am working on bellhop model
my profile is
bathy = [
    [0, 1],    # 30 m water depth at the transmitter
    [100, 3500],  # 20 m water depth 300 m away
    [500, 3700],  # 20 m water depth 300 m away
    [1000, 4000], # 25 m water depth at 1 km
    [2000, 4200],  # 25 m water depth at 2 km
    [3000, 4500],  # 25 m water depth at 2 km
    [4000, 4750],  # 25 m water depth at 2 km
    [5000, 5000],  # 25 m water depth at 2 km
    [10000, 5200],  # 25 m water depth at 2 km
    [100000, 5200]  # 25 m water depth at 2 km
]
ssp = [
    [ 0, 1542],  # 1540 m/s at the surface
    [100, 1527],  # 1530 m/s at 10 m depth
    [200, 1508],  # 1532 m/s at 20 m depth
    [300, 1499],  # 1533 m/s at 25 m depth
    [400, 1495],   # 1535 m/s at the seabed
    [500, 1492],  # 1540 m/s at the surface
    [600, 1490],  # 1530 m/s at 10 m depth
    [700, 1489],  # 1532 m/s at 20 m depth
    [800, 1487],  # 1533 m/s at 25 m depth
    [900, 1487],   # 1535 m/s at the seabed
    [1000, 1487],  # 1540 m/s at the surface
    [1100, 1487],  # 1530 m/s at 10 m depth
    [1200, 1487],  # 1532 m/s at 20 m depth
    [1300, 1487],  # 1533 m/s at 25 m depth
    [1400, 1487],   # 1535 m/s at the seabed
    [1500, 1487],  # 1540 m/s at the surface
    [1600, 1488],  # 1530 m/s at 10 m depth
    [1700, 1488],  # 1532 m/s at 20 m depth
    [1800, 1489],  # 1533 m/s at 25 m depth
    [1900, 1490],  # 1535 m/s at the seabed
    [2000, 1491],  # 1540 m/s at the surface
    [2100, 1492],  # 1530 m/s at 10 m depth
    [2200, 1493],  # 1532 m/s at 20 m depth
    [2300, 1494],  # 1533 m/s at 25 m depth
    [2400, 1496],  # 1535 m/s at the seabed
    [2500, 1497],  # 1540 m/s at the surface
    [2600, 1498],  # 1530 m/s at 10 m depth
    [2700, 1500],  # 1532 m/s at 20 m depth
    [2800, 1501],  # 1533 m/s at 25 m depth
    [2900, 1503],   # 1535 m/s at the seabed
    [3000, 1504],  # 1540 m/s at the surface
    [3100, 1506],  # 1530 m/s at 10 m depth
    [3200, 1507],  # 1532 m/s at 20 m depth
    [3300, 1509],  # 1533 m/s at 25 m depth
    [3400, 1511],   # 1535 m/s at the seabed
    [3500, 1512],  # 1540 m/s at the surface
    [3600, 1514],  # 1530 m/s at 10 m depth
    [3700, 1516],  # 1532 m/s at 20 m depth
    [3800, 1517],  # 1533 m/s at 25 m depth
    [3900, 1519],   #
    [4000, 1521],  # 1540 m/s at the surface
    [4100, 1523],  # 1530 m/s at 10 m depth
    [4200, 1524],  # 1532 m/s at 20 m depth
    [4300, 1526],  # 1533 m/s at 25 m depth
    [4400, 1528],   # 1535 m/s at the seabed
    [4500, 1530],  # 1540 m/s at the surface
    [4600, 1532],  # 1530 m/s at 10 m depth
    [4700, 1534],  # 1532 m/s at 20 m depth
    [4800, 1535],  # 1533 m/s at 25 m depth
    [4900, 1537],  #
    [5000, 1539],  # 1540 m/s at the surface
    [5100, 1541],  # 1530 m/s at 10 m depth
    [5400, 1543]  # 1532 m/s at 20 m depth
]

env = pm.create_env2d(
     bottom_absorption = 0.1,
      bottom_density = 1600,
    bottom_roughness = 0,
   bottom_soundspeed = 1600,
               depth =bathy,
        depth_interp = "curvilinear",
           frequency = 50,
           max_angle = 80,
           min_angle = -80,
              nbeams = 5000,
            rx_depth = 10,
            rx_range = 100000,
          soundspeed = ssp,
   #soundspeed_interp = spline,
             surface = None,
      #surface_interp = linear,
            tx_depth = 500,
   tx_directionality = None
   
)
while calculating rays I am getting this error

import arlpy.uwapm as pm

rays = pm.compute_eigenrays(env, debug=True)
pm.plot_rays(rays, env=env, width=900)

error:

[DEBUG] Model: bellhop
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [25], in <cell line: 3>()
      1 import arlpy.uwapm as pm
      2 #env = pm.create_env2d()
----> 3 rays = pm.compute_eigenrays(env, debug=True)
      4 pm.plot_rays(rays, env=env, width=900)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\arlpy\uwapm.py:354, in compute_eigenrays(env, tx_depth_ndx, rx_depth_ndx, rx_range_ndx, model, debug)
    352 if debug:
    353     print('[DEBUG] Model: '+model_name)
--> 354 return model.run(env, eigenrays, debug)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\arlpy\uwapm.py:607, in _Bellhop.run(self, env, task, debug)
    605 else:
    606     try:
--> 607         results = taskmap[task][1](fname_base)
    608     except FileNotFoundError:
    609         print('[WARN] Bellhop did not generate expected output file')

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\arlpy\uwapm.py:821, in _Bellhop._load_rays(self, fname_base)
    814             ray[k,:] = self._readf(f, (float, float))
    815         rays.append(_pd.DataFrame({
    816             'angle_of_departure': [a],
    817             'surface_bounces': [sb],
    818             'bottom_bounces': [bb],
    819             'ray': [ray]
    820         }))
--> 821 return _pd.concat(rays)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pandas\util\_decorators.py:311, in deprecate_nonkeyword_arguments.<locals>.decorate.<locals>.wrapper(*args, **kwargs)
    305 if len(args) > num_allow_args:
    306     warnings.warn(
    307         msg.format(arguments=arguments),
    308         FutureWarning,
    309         stacklevel=stacklevel,
    310     )
--> 311 return func(*args, **kwargs)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pandas\core\reshape\concat.py:294, in concat(objs, axis, join, ignore_index, keys, levels, names, verify_integrity, sort, copy)
     90 @deprecate_nonkeyword_arguments(version=None, allowed_args=["objs"])
     91 def concat(
     92     objs: Iterable[NDFrame] | Mapping[Hashable, NDFrame],
   (...)
    101     copy: bool = True,
    102 ) -> FrameOrSeriesUnion:
    103     """
    104     Concatenate pandas objects along a particular axis with optional set logic
    105     along the other axes.
   (...)
    292     ValueError: Indexes have overlapping values: ['a']
    293     """
--> 294     op = _Concatenator(
    295         objs,
    296         axis=axis,
    297         ignore_index=ignore_index,
    298         join=join,
    299         keys=keys,
    300         levels=levels,
    301         names=names,
    302         verify_integrity=verify_integrity,
    303         copy=copy,
    304         sort=sort,
    305     )
    307     return op.get_result()

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pandas\core\reshape\concat.py:351, in _Concatenator.__init__(self, objs, axis, join, keys, levels, names, ignore_index, verify_integrity, copy, sort)
    348     objs = list(objs)
    350 if len(objs) == 0:
--> 351     raise ValueError("No objects to concatenate")
    353 if keys is None:
    354     objs = list(com.not_none(*objs))

ValueError: No objects to concatenate

mchitre, 2023-01-04:
I believe BELLHOP needs your SSP maximum depth to be equal to the bathymetry maximum depth. Try with that and see what you get. If problem persists, I'd suggest looking at the debug files (BELLHOP inputs and outputs) to see what might be wrong. You could post them here, if you need help looking at them.

heeradshakeera, 2023-01-16:
> I believe BELLHOP needs your SSP maximum depth to be equal to the bathymetry maximum depth. Try with that and see what you get. If problem persists, I'd suggest looking at the debug files (BELLHOP [...]

I changed shower it showing same error:

bathy = [
    [0, 1],    # 30 m water depth at the transmitter
    [100, 3500],  # 20 m water depth 300 m away
    [500, 3700],  # 20 m water depth 300 m away
    [1000, 4000], # 25 m water depth at 1 km
    [2000, 4200],  # 25 m water depth at 2 km
    [3000, 4500],  # 25 m water depth at 2 km
    [4000, 4750],  # 25 m water depth at 2 km
    [5000, 5000],  # 25 m water depth at 2 km
    [10000, 5200],  # 25 m water depth at 2 km
    [100000, 5500]  # 25 m water depth at 2 km
]

ssp = [
    [ 0, 1542],  # 1540 m/s at the surface
    [500, 1492],  # 1530 m/s at 10 m depth
    [1000, 1489],  # 1532 m/s at 20 m depth
    [1500, 1488],  # 1533 m/s at 25 m depth
    [2000, 1491],   # 1535 m/s at the seabed
    [2500, 1497],  # 1540 m/s at the surface
    [3000, 1504],  # 1530 m/s at 10 m depth
    [3500, 1512],  # 1532 m/s at 20 m depth
    [4000, 1521],  # 1533 m/s at 25 m depth
    [4500, 1530],   # 1535 m/s at the seabed
    [5000, 1539],  # 1540 m/s at the surface
    [5500, 1542],  # 1530 m/s at 10 m depth
    
]
env = pm.create_env2d(
     bottom_absorption = 0.1,
      bottom_density = 1600,
    bottom_roughness = 0,
   bottom_soundspeed = 1600,
               depth =bathy,
        depth_interp = "curvilinear",
           frequency = 50,
           max_angle = 80,
           min_angle = -80,
              nbeams = 5000,
            rx_depth = 10,
            rx_range = 100000,
          soundspeed = ssp,
   #soundspeed_interp = spline,
             surface = None,
      #surface_interp = linear,
            tx_depth = 500,
   tx_directionality = None
   
)

ValueError                                Traceback (most recent call last)
Input In [58], in <cell line: 1>()
----> 1 rays = pm.compute_eigenrays(env)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\arlpy\uwapm.py:354, in compute_eigenrays(env, tx_depth_ndx, rx_depth_ndx, rx_range_ndx, model, debug)
    352 if debug:
    353     print('[DEBUG] Model: '+model_name)
--> 354 return model.run(env, eigenrays, debug)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\arlpy\uwapm.py:607, in _Bellhop.run(self, env, task, debug)
    605 else:
    606     try:
--> 607         results = taskmap[task][1](fname_base)
    608     except FileNotFoundError:
    609         print('[WARN] Bellhop did not generate expected output file')

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\arlpy\uwapm.py:821, in _Bellhop._load_rays(self, fname_base)
    814             ray[k,:] = self._readf(f, (float, float))
    815         rays.append(_pd.DataFrame({
    816             'angle_of_departure': [a],
    817             'surface_bounces': [sb],
    818             'bottom_bounces': [bb],
    819             'ray': [ray]
    820         }))
--> 821 return _pd.concat(rays)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pandas\util\_decorators.py:311, in deprecate_nonkeyword_arguments.<locals>.decorate.<locals>.wrapper(*args, **kwargs)
    305 if len(args) > num_allow_args:
    306     warnings.warn(
    307         msg.format(arguments=arguments),
    308         FutureWarning,
    309         stacklevel=stacklevel,
    310     )
--> 311 return func(*args, **kwargs)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pandas\core\reshape\concat.py:294, in concat(objs, axis, join, ignore_index, keys, levels, names, verify_integrity, sort, copy)
     90 @deprecate_nonkeyword_arguments(version=None, allowed_args=["objs"])
     91 def concat(
     92     objs: Iterable[NDFrame] | Mapping[Hashable, NDFrame],
   (...)
    101     copy: bool = True,
    102 ) -> FrameOrSeriesUnion:
    103     """
    104     Concatenate pandas objects along a particular axis with optional set logic
    105     along the other axes.
   (...)
    292     ValueError: Indexes have overlapping values: ['a']
    293     """
--> 294     op = _Concatenator(
    295         objs,
    296         axis=axis,
    297         ignore_index=ignore_index,
    298         join=join,
    299         keys=keys,
    300         levels=levels,
    301         names=names,
    302         verify_integrity=verify_integrity,
    303         copy=copy,
    304         sort=sort,
    305     )
    307     return op.get_result()

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pandas\core\reshape\concat.py:351, in _Concatenator.__init__(self, objs, axis, join, keys, levels, names, ignore_index, verify_integrity, copy, sort)
    348     objs = list(objs)
    350 if len(objs) == 0:
--> 351     raise ValueError("No objects to concatenate")
    353 if keys is None:
    354     objs = list(com.not_none(*objs))

ValueError: No objects to concatenate

mchitre, 2023-01-17:
Did you look at the input/output files from BELLHOP (in debug mode)? Please post those, as they're likely to tell you where the problem lies.

heeradshakeera, 2023-01-17:
How to check those.

mchitre, 2023-01-24:
The problem seems to be that your SSP is defined to 5400.0 m but your bathymetry never reaches that depth. Bellhop doesn't seem to like that. I played with changing bathymetry to be flat or sloping with one end being at 5400.0 m, and it works fine.

heeradshakeera, 2023-01-25:
Thank you can you please post the bathymetry which worked for u for this application.

mchitre, 2023-01-25:
This works:
```
bathy = [
    [0, 3500],
    [100, 3500],
    [500, 3700],
    [1000, 4000],
    [2000, 4200],
    [3000, 4500],
    [4000, 4750],
    [5000, 5000],
    [10000, 5200],
    [100000, 5200]
]
```
Just changed the first entry to be more than 500 m, where your source is. Otherwise the source is in the seabed.

mchitre, 2024-01-29:
I am assuming this was resolved.

closed 2024-01-29 by mchitre