org-arl/arlpy #79: arlpy outputs debug messages to stdout

vlaskine, 2023-03-29:
Various arlpy methods output debug messages to stdout. It makes it impossible to use arlpy in the standard Linux pipeline mode like **cat some-input.csv | some-script-using-arlpy | process-output** ... It is quite a limitation and inconvenience. The fix is really straightforward.

The places I tripped over are (there may be other cases):

- using **print( something )**: should be **print( something, file=sys.stderr )** or **logging.info(...)** or alike.
- uwapm.py: **_proc.call(['bellhop.exe'] + list(args), stderr=_proc.STDOUT)** should be **stderr=_proc.STDERR**

mchitre, 2023-04-02:
Fair enough. Would you be willing to PR the changes needed @vlaskine?

vlaskine, 2023-04-04:
Thanks, I just created a pull request: https://github.com/org-arl/arlpy/pull/80

It fixes only uwapm, because that's the only part of uwapy that we are currently using and thus that's the only one that I could test.

mchitre, 2023-04-04:
Fixed by #80

closed 2023-04-04 by mchitre