org-arl/arlpy #81: ARRFile output in two lines instead of one

IvanaEscobar, 2023-04-05:
## Previous behavior: 

ARRFile printed output on a single line in the format:
```
Amplitude       phase          delay         delay(imag)     start angle     end angle      bounce top   bounce bottom
7.2071263E-05   1862.231       9.524935      0.0000000E+00   22.00000        -54.49823      7            8
```

`uwapm.py` provides a way to read data on a single line as `data`:
https://github.com/org-arl/arlpy/blob/003b3111ab9aa17666af7af71e3b015734f028c1/arlpy/uwapm.py#L782-L798

## Issue: 

ARRFile prints output over multiple lines, with end angle, top, and bottom bounces on a second line:
```
Amplitude       phase          delay         delay(imag)     start angle     
7.2071263E-05   1862.231       9.524935      0.0000000E+00   22.00000        
end angle      bounce top   bounce bottom
-54.49823      7            8
```

[Sample arrival output](https://github.com/IvanaEscobar/oceanAcouPy/blob/main/example.arr) compiled in a Linux-GNU x86_64 machine with `ifort (IFORT) 180.0 20170811`. 

## Personal resolution: 
In `uwapm.py`, read in two lines at a time for arrivals data. 
https://github.com/IvanaEscobar/arlpy/blob/5ba7be0e4ef0f27ea9dca74d2c6ac5ccfb991329/arlpy/uwapm.py#L779-L796

mchitre, 2023-04-09:
@IvanaEscobar, the personal resolution you have will fail with older versions of Bellhop, I presume?

IvanaEscobar, 2023-04-09:
I don’t have access to older versions of Bellhop, but if you do please send me the files and I’ll check.

Ivana

mchitre, 2023-04-13:
@IvanaEscobar 
http://oalib.hlsresearch.com/AcousticsToolbox/at_2020_11_4.zip

IvanaEscobar, 2023-04-13:
That's the version I am running now. So I'm wondering I'll check in a blank shell environment to see if it's an artifact of my environment or a BELLHOP feature.

mchitre, 2023-04-14:
I tested with `v2022_4_20` and got the format we expect (single line):
```
 '2D'
   50.000000000000000
           1   5.00000000
           1   10.0000000
           1   500.000000
          14
          14
   3.69561130E-05   691.611511      0.330679715      -6.71795817E-07  -10.8163252       10.8163252               3           2
   1.55016805E-05   694.596619      0.330678821      -6.71793998E-07  -10.6122437       10.6122437               3           2
   7.48764505E-05   547.096558      0.328502387      -6.67372433E-07  -8.57142735      -8.57142735               2           2
   1.98099733E-05   550.605835      0.328501105      -6.67369818E-07  -8.36734581      -8.36734581               2           2
   6.28456473E-04   474.043152      0.326827168      -6.63969161E-07  -6.27594423       6.27594423               2           1
   9.51317779E-04   316.661926      0.325662494      -6.61603053E-07  -4.00235319      -4.00235319               1           1
   1.99911278E-03   180.000000      0.325013697      -6.60284968E-07  -1.71835613       1.71835613               1           0
   1.99989555E-03   0.00000000      0.324883938      -6.60021328E-07  0.572938502      0.572938502               0           0
   1.17621513E-03   149.518082      0.325273782      -6.60813384E-07   2.86221385      -2.86221385               0           1
   7.71005871E-04   306.181641      0.326180756      -6.62655907E-07   5.14157057       5.14157057               1           1
   9.72367125E-05   388.823822      0.327601463      -6.65542188E-07   7.34693813      -7.34693813               1           2
   3.78462682E-05   385.092896      0.327600598      -6.65540426E-07   7.55101967      -7.55101967               1           2
   5.07089171E-05   530.249878      0.329528779      -6.69457620E-07   9.59183598       9.59183598               2           2
   1.82773365E-05   527.022827      0.329527825      -6.69455687E-07   9.79591751       9.79591751               2           2
```

IvanaEscobar, 2023-04-16:
@mchitre I figured out the underlying issue caused by compilers. Credit to [this thread](https://stackoverflow.com/questions/17407232/avoid-newline-in-list-directed-output-with-intel-fortran-compiler). 

I adjusted my compiling flags by adding `-no-wrap-margin`, and was able to avoid a two lined output. Do you know of a way to officially contribute to the Acoustics Toolbox? This update would be useful for other Intel compiler users.

IvanaEscobar, 2023-04-16:
P.S. Also checked with the newest version of Bellhop and the ARRFile has not changed its format

closed 2023-04-16 by IvanaEscobar

mchitre, 2023-04-17:
Ah, makes sense, compiler quirks! :-)

You may want to get in touch with Michael Porter, who is the author of OALIB to contribute.