org-arl/arlpy #84: Compatibility issues with Bokeh version 3.1.1

patel999jay, 2023-06-07:
Hello,

I've been using the arlpy toolbox and recently updated Bokeh from version 2.x.x to 3.1.1. After the update, I've encountered some issues with the plotting functions in arlpy.

When running `pm.plot_env(env, width=900)`, I get the following error:

**AttributeError: unexpected attribute 'plot_width' to figure, similar attributes are outer_width, width or min_width**

This suggests that the attribute 'plot_width' is no longer recognized by the Bokeh Figure object in version 3.1.1. Instead, the 'width' attribute should be used.

When running `rays = pm.compute_eigenrays(env , debug=True)` and `pm.plot_rays(rays, env=env, width=900)`, I get this error:

**ValueError: failed to validate FigureOptions(...).x_range: expected an element of either Instance(Range), Either(Tuple(Float, Float), Tuple(Datetime, Datetime), Tuple(TimeDelta, TimeDelta)), Seq(String), Object(Series) or Object(GroupBy), got None**

This error suggests that the 'x_range' attribute of the Bokeh Figure object is not being set correctly in the arlpy plotting functions.

These issues seem to be related to changes in the Bokeh API between version 2.x.x and 3.1.1. To resolve these, the arlpy library would need to be updated to use the new Bokeh API. This would involve updating any instances where Bokeh figures are created or modified in the arlpy code.

Please note that these changes could **potentially break compatibility with older versions of Bokeh.** Therefore, it might be necessary to implement some sort of version checking to ensure that the correct API is used depending on the installed version of Bokeh.

patel999jay, 2023-06-12:
@mchitre : i reviewed the issue, and created a PR, ready to merge.

bbbrandyn, 2023-06-14:
Hi, I also had this issue and also had to change `plot_height=height` to 'height=height`

Did you have this too @patel999jay ?

mchitre, 2023-06-15:
> Hi, I also had this issue and also had to change `plot_height=height` to `height=height`

Yes, see https://github.com/org-arl/arlpy/blob/acb1b7abc815885f8825d8ca2f1910d44ecd0086/arlpy/plot.py#L61

PR #85 (2023-06-14): Fixes org-arl/arlpy#84
I had tested this with newer python version `python=3.11` and it looks like the issue has been resolved with these changes.

This PR updates the arlpy library to ensure compatibility with Bokeh version 3.1.1 and the latest version of Scipy version 1.10.1. The changes include:

Replaced `plot_width` with `width` and `plot_height` with `height` in all instances where Bokeh figures are created or modified. This change was necessary due to updates in the Bokeh API between version 2.x.x and 3.1.1. In the newer version, the `plot_width` and `plot_height` attributes have been r [... cut]