org-arl/arlpy #86: Ability to plot a transmission loss slice at one receiver depth

bbbrandyn, 2023-06-21:
I was trying to implement in arlpy a way to plot a transmission loss slice at one receiver depth such as is done by the plottlr.m script in the Acoustics Toolbox. But I noticed that the values in each row of a `tloss` are complex numbers so I'm not sure how to simply plot the transmission loss at one depth. How can I use the data returned from `compute_transmission_loss()` in this way? This image is what I'm trying to accomplish for a selected receiver depth.
[image]

mchitre, 2023-07-30:
See #72 and #83 for a discussion on complex transmission loss. With #83, the transmission loss will be reported as real, with the phase compensated in the arrival time.

mchitre, 2023-07-30:
Released v1.8.3, which should fix this.

closed 2023-08-05 by mchitre