org-arl/arlpy #89: Arrival Amplitude Calculation raises error

jamesmr13, 2023-08-30:
Hello,

My Specs:

Windows10 Python 3.11
arlpy version: 1.8.3
Acoustic Toolbox: Windows binaries updated 20201102

I received the following error when calling compute_arrivals() on the default 2D environment.

```
[DEBUG] Model: bellhop
Traceback (most recent call last):
  File "c:\Users\James\Documents\Testing\arlpy\test.py", line 10, in <module>
    arrivals = pm.compute_arrivals(env, debug=True)
  File "C:\Users\James\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\arlpy\uwapm.py", line 325, in compute_arrivals
    return model.run(env, arrivals, debug)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\James\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\arlpy\uwapm.py", line 607, in run
    results = taskmap[task][1](fname_base)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "C:\Users\James\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\arlpy\uwapm.py", line 793, in _load_arrivals
    'arrival_amplitude': [data[0] * _np.exp( -1j * (_np.deg2rad(data[1]) + freq * 2 * _np.pi * (data[3] * 1j +  data[2])))],
                                                                           ~~~~~~~~~^~~~~~~~
TypeError: can't multiply sequence by non-int of type 'float'
```

I opened the source file uwapm.py and uncommented the line above where the error occurs and commented the line raising the error as shown here:

```p
'arrival_amplitude': [data[0]*_np.exp(1j * data[1]* _np.pi/180)],
#'arrival_amplitude': [data[0] * _np.exp( -1j * (_np.deg2rad(data[1]) + freq * 2 * _np.pi * (data[3] * 1j +  data[2])))],
```

This resolves the error and produces the arrivals expected from the Bellhop notebook example it seems that the freq variable is a string based on the TypeError.

mchitre, 2023-09-02:
Can't see a reason for `freq` to be a string. Could be specific to versions of BELLHOP and platform. @jamesmr13 Could you post the `.arr` file by running a model with `debug=True` and copying the generated file out? That'll make it easier to track the error.

jamesmr13, 2023-09-02:
Absolutely, here is the .arr file

```
 '2D'
   25000.000000000000     
           1   5.00000000    
           1   150.000000    
           1   25000.0000    
          11
          11
   5.97442693E-07   563.183228       21.1272297      -1.44080797E-04  -37.5755997       38.0623932               3           2
   1.04485480E-07   383.583649       21.0079098      -1.43241370E-04  -37.1238976      -37.6155090               2           2
   5.10202142E-07   383.584167       21.0079098      -1.43241356E-04  -37.1232986      -37.6149216               2           2
   7.02727903E-06   391.565857       17.7458286      -1.21098521E-04  -19.5497570       20.4374142               2           1
   4.80200652E-06   391.567627       17.7458305      -1.21098492E-04  -19.5491562       20.4368172               2           1
   4.87028228E-06   213.902969       17.6792793      -1.20602490E-04  -18.8165512      -19.9885330               1           1
   7.87280896E-06   213.905029       17.6792793      -1.20602468E-04  -18.8159504      -19.9879894               1           1
   1.05410163E-05   33.9947395       17.6772060      -1.20586046E-04   18.7899494      -19.9643593               0           1
   2.23987695E-06   33.9926682       17.6772060      -1.20586075E-04   18.7905502      -19.9649029               0           1
   3.74889191E-06   211.643051       17.7436752      -1.21081655E-04   19.5237560       20.4114590               1           1
   8.11374775E-06   211.641281       17.7436752      -1.21081677E-04   19.5243568       20.4120579               1           1
```

I too can't figure out why freq would be a string looking at the _readf function of uwapm.py it seems like the call to the function in line 758 of uwapm.py correctly specifies that a float should be returned. I even tried to alter that functions call to the following line:

```python
freq = self._readf(f, (float,), float)
```

in order to specify the dtype parameter as well and I still get same error.

jamesmr13, 2023-09-02:
Another update:

I called 

```python
type(freq)
```

In the code and it printed <class 'tuple'>

jamesmr13, 2023-09-04:
So I was able to correctly use the
```python
'arrival_amplitude': [data[0] * _np.exp( -1j * (_np.deg2rad(data[1]) + freq[0] * 2 * _np.pi * (data[3] * 1j +  data[2])))],
```

since the _readf function is returning a tuple of size 1 I simply accessed the first element of the tuple and the same .arr file I pasted above generated the arrivals DataFrame below:

```
[DEBUG] Bellhop working files: C:\Users\jrober23\AppData\Local\Temp\tmpgki780_v.*
    tx_depth_ndx  rx_depth_ndx  rx_range_ndx  tx_depth  rx_depth  rx_range  ...  time_of_arrival  complex_time_of_arrival  angle_of_departure  angle_of_arrival  surface_bounces  bottom_bounces
1              0             0             0       5.0     150.0   25000.0  ...        21.127230      21.127230-0.000144j          -37.575600         38.062393                3               2
2              0             0             0       5.0     150.0   25000.0  ...        21.007910      21.007910-0.000143j          -37.123898        -37.615509                2               2
3              0             0             0       5.0     150.0   25000.0  ...        21.007910      21.007910-0.000143j          -37.123299        -37.614922                2               2
4              0             0             0       5.0     150.0   25000.0  ...        17.745829      17.745829-0.000121j          -19.549757         20.437414                2               1
5              0             0             0       5.0     150.0   25000.0  ...        17.745831      17.745831-0.000121j          -19.549156         20.436817                2               1
6              0             0             0       5.0     150.0   25000.0  ...        17.679279      17.679279-0.000121j          -18.816551        -19.988533                1               1
7              0             0             0       5.0     150.0   25000.0  ...        17.679279      17.679279-0.000121j          -18.815950        -19.987989                1               1
8              0             0             0       5.0     150.0   25000.0  ...        17.677206      17.677206-0.000121j           18.789949        -19.964359                0               1
9              0             0             0       5.0     150.0   25000.0  ...        17.677206      17.677206-0.000121j           18.790550        -19.964903                0               1
10             0             0             0       5.0     150.0   25000.0  ...        17.743675      17.743675-0.000121j           19.523756         20.411459                1               1
11             0             0             0       5.0     150.0   25000.0  ...        17.743675      17.743675-0.000121j           19.524357         20.412058                1               1

[11 rows x 14 columns]
8
```

MikJaw94, 2023-09-07:
I have the same issue. I have not upgraded the bellhop version, but I upgraded from python 3.9 to 3.11.

Any workarounds?

jamesmr13, 2023-09-07:
> I have the same issue. I have not upgraded the bellhop version, but I upgraded from python 3.9 to 3.11.
> 
> Any workarounds?
> 
> Edit:
> It was running without issues a couple of months ago.

If you edit uwapm.py by changing freq to freq[0] it should work

mchitre, 2023-09-10:
Thanks @jamesmr13 for the detailed analysis. Seems you have a quick fix for now, so this is hopefully not blocking. I'm out at experiments this week, but should have a chance to find the root cause and fix it in a couple of weeks.

mchitre, 2023-10-23:
@patel999jay https://github.com/org-arl/arlpy/issues/72#issuecomment-1769435948 is the same issue right?

patel999jay, 2023-10-23:
Yeah i guess same issue.

commit 21bcf496 (2023-10-31): Fixes #89

Fixes arrival amplitude calculations
closed 2023-10-31 by mchitre