org-arl/arlpy #90: failed to validate FigureOptions(...).y_range

blackknifes, 2023-09-04:
# Versions
- Bellhop: 2018
- arlpy: 1.8.3

# My source code
```
import arlpy.uwapm as pm
import arlpy.plot as plt
import numpy as np

print(pm.models())
env = pm.create_env2d()
pm.print_env(env)

pm.plot_env(env, width=900)

rays = pm.compute_eigenrays(env, debug=True)
pm.plot_rays(rays, env=env, width=900)

arrivals = pm.compute_arrivals(env)
pm.plot_arrivals(arrivals, width=900)

arrivals[arrivals.arrival_number < 10][['time_of_arrival', 'angle_of_arrival', 'surface_bounces', 'bottom_bounces']]

ir = pm.arrivals_to_impulse_response(arrivals, fs=96000)
plt.plot(np.abs(ir), fs=96000, width=900)

bathy = [
    [0, 30],    # 30 m water depth at the transmitter
    [300, 20],  # 20 m water depth 300 m away
    [1000, 25]  # 25 m water depth at 1 km
]

ssp = [
    [ 0, 1540],  # 1540 m/s at the surface
    [10, 1530],  # 1530 m/s at 10 m depth
    [20, 1532],  # 1532 m/s at 20 m depth
    [25, 1533],  # 1533 m/s at 25 m depth
    [30, 1535]   # 1535 m/s at the seabed
]

env = pm.create_env2d(
    depth=bathy,
    soundspeed=ssp,
    bottom_soundspeed=1450,
    bottom_density=1200,
    bottom_absorption=1.0,
    tx_depth=15
)

pm.print_env(env)

pm.plot_env(env, width=900)

pm.plot_ssp(env)

rays = pm.compute_eigenrays(env)
pm.plot_rays(rays, env=env, width=900)

rays = pm.compute_rays(env)
pm.plot_rays(rays, env=env, width=900)

env['rx_range'] = np.linspace(0, 1000, 1001)
env['rx_depth'] = np.linspace(0, 30, 301)

tloss = pm.compute_transmission_loss(env)
pm.plot_transmission_loss(tloss, env=env, clim=[-60,-30], width=900)

tloss = pm.compute_transmission_loss(env, mode='incoherent')
pm.plot_transmission_loss(tloss, env=env, clim=[-60,-30], width=900)

beampattern = np.array([
    [-180,  10], [-170, -10], [-160,   0], [-150, -20], [-140, -10], [-130, -30],
    [-120, -20], [-110, -40], [-100, -30], [-90 , -50], [-80 , -30], [-70 , -40],
    [-60 , -20], [-50 , -30], [-40 , -10], [-30 , -20], [-20 ,   0], [-10 , -10],
    [  0 ,  10], [ 10 , -10], [ 20 ,   0], [ 30 , -20], [ 40 , -10], [ 50 , -30],
    [ 60 , -20], [ 70 , -40], [ 80 , -30], [ 90 , -50], [100 , -30], [110 , -40],
    [120 , -20], [130 , -30], [140 , -10], [150 , -20], [160 ,   0], [170 , -10],
    [180 ,  10]
])
env['tx_directionality'] = beampattern

tloss = pm.compute_transmission_loss(env)
pm.plot_transmission_loss(tloss, env=env, clim=[-60,-30], width=900)

surface = np.array([[r, 0.5+0.5*np.sin(2*np.pi*0.005*r)] for r in np.linspace(0,1000,1001)])
env['surface'] = surface

tloss = pm.compute_transmission_loss(env)
pm.plot_transmission_loss(tloss, env=env, clim=[-60,-30], width=900)

env['rx_range'] = 800
env['rx_depth'] = 15

rays = pm.compute_eigenrays(env)
pm.plot_rays(rays, env=env, width=900)

arrivals = pm.compute_arrivals(env)
pm.plot_arrivals(arrivals, dB=True, width=900)
```

# Exception
```
ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
failed to validate FigureOptions(...).y_range: expected an element of either Instance(Range), Either(Tuple(Float, Float), Tuple(Datetime, Datetime), Tuple(TimeDelta, TimeDelta)), Seq(String), Object(Series) or Object(GroupBy), got None
  File "C:\Python\Python39\Lib\site-packages\bokeh\core\property\bases.py", line 364, in prepare_value
    raise ValueError(f"failed to validate {obj_repr}.{name}: {error}")
  File "C:\Python\Python39\Lib\site-packages\bokeh\core\property\descriptors.py", line 332, in __set__
    value = self.property.prepare_value(obj, self.name, value)
  File "C:\Python\Python39\Lib\site-packages\bokeh\core\has_props.py", line 327, in __setattr__
    return super().__setattr__(name, value)
  File "C:\Python\Python39\Lib\site-packages\bokeh\core\has_props.py", line 295, in __init__
    setattr(self, name, value)
  File "C:\Python\Python39\Lib\site-packages\bokeh\util\options.py", line 75, in __init__
    super().__init__(**props)
  File "C:\Python\Python39\Lib\site-packages\bokeh\plotting\_figure.py", line 187, in __init__
    opts = FigureOptions(kw)
  File "C:\Python\Python39\Lib\site-packages\arlpy\plot.py", line 61, in _new_figure
    f = _bplt.figure(title=title, width=width, height=height, x_range=xlim, y_range=ylim, x_axis_label=xlabel, y_axis_label=ylabel, x_axis_type=xtype, y_axis_type=ytype, tools=tools)
  File "C:\Python\Python39\Lib\site-packages\arlpy\plot.py", line 368, in plot
    _figure = _new_figure(title, width, height, xlabel, ylabel, xlim, ylim, xtype, ytype, interactive)
  File "C:\Python\Python39\Lib\site-packages\arlpy\uwapm.py", line 496, in plot_rays
    _plt.plot(row.ray[:,0]/divisor, -row.ray[:,1], color=c, xlabel=xlabel, ylabel='Depth (m)', **kwargs)
  File "D:\lkq\python\arlpy_demo\arlpy_demo.py", line 12, in <module>
    pm.plot_rays(rays, env=env, width=900)
  File "C:\Python\Python39\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Python\Python39\Lib\runpy.py", line 197, in _run_module_as_main (Current frame)
    return _run_code(code, main_globals, None,
ValueError: failed to validate FigureOptions(...).y_range: expected an element of either Instance(Range), Either(Tuple(Float, Float), Tuple(Datetime, Datetime), Tuple(TimeDelta, TimeDelta)), Seq(String), Object(Series) or Object(GroupBy), got None
```

mchitre, 2023-09-10:
Sounds like a problem with Bokeh plotting. What version of Python and Bokeh are you on?

notthetup, 2023-09-21:
Also seeing this on `Python 3.11` and `bokeh==3.0.3` and also on `bokeh=3.2.2`

PR #91 (2023-09-21): Fixing ploting functionality with newer version of bokeh
Fix #90
commit cddcbdca (2023-09-21): fix(plot): fixing ploting functionality with newer version of bokeh. fix #90
closed 2023-09-21 by mchitre