org-arl/arlpy #95: compute_arrivals and compute_eigenrays do not work in parallel

SquirrelKnight, 2023-11-10:
Hello,

I am currently working on a project to derive raypaths for thousands of receiver/transmitter pairs (between an autonomous vehicle and a seafloor transponder). Because of this, I was hoping to implement parallel functionality using a fairly simple script and joblib. Unfortunately, it seems that Bellhop is unable to deal with this. I get the following messages.

> [BELLHOP]  *** FATAL ERROR ***
[BELLHOP]  Generated by program or subroutine: BELLHOP - READIN
[BELLHOP]  Unable to open the environmental file
[BELLHOP] 

When running my script in series, it functions just fine.

Parallelization compatibility would be an excellent addition.

SquirrelKnight, 2023-11-10:
I have discovered that the issue was with joblib. Apparently joblib is not able to access the temporary directory for parallel processing: https://github.com/joblib/joblib/issues/391

Dask parallel processing works great, however.

closed 2023-11-10 by SquirrelKnight

mchitre, 2023-11-11:
Glad to know that this was resolved.