org-arl/arlpy #101: Not able to compute or plot arrival times/ rays using bellhop model in python

vtzirakis, 2024-03-25:
Hello!

I'm trying to use python to simulate marine environments and compute rays, arrival times etc, as I am not familiar with matlab. All works fine ( I can set and plot  the environment), however when I try to compute the rays or arrival times ( arrivals = pm.compute_arrivals(env) ) I get the following message in jupyter: 

[WARN] Bellhop did not generate expected output file. 

As a result I can not plot the rays or arrival times, as the variable values are probably set to Nan or something. Specifically, when I try to plot arrival times I get this error: 
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[16], line 2
      1 arrivals = pm.compute_arrivals(env)
----> 2 pm.plot_rays(arrivals, env=env, width=900)

File ~\AppData\Roaming\Python\Python311\site-packages\arlpy\uwapm.py:478, in plot_rays(rays, env, invert_colors, **kwargs)
    461 def plot_rays(rays, env=None, invert_colors=False, **kwargs):
    462     """Plots ray paths.
    463 
    464     :param rays: ray paths
   (...)
    476     >>> pm.plot_rays(rays, width=1000)
    477     """
--> 478     rays = rays.sort_values('bottom_bounces', ascending=False)
    479     max_amp = _np.max(_np.abs(rays.bottom_bounces)) if len(rays.bottom_bounces) > 0 else 0
    480     if max_amp <= 0:

AttributeError: 'NoneType' object has no attribute 'sort_values' 

I do not know what the problem is and since there is not much information online about bellhop in python, I am stuck. Any help would be really appriciated. Thanks in advance!

Vassilis

mchitre, 2024-03-28:
Run your code with `debug=True` and copy out the generated files. Looking at their content might give you a hint as to what is wrong.

Arthanor, 2024-11-26:
Hi, I assume the previous poster managed to find their issue and, instead of opening a new one, I figured this is quite appropriate since I also cannot generate rays (in the domain I need). For context, here is a figure for the rays with reduced source angle so that the results are limited to what worked:
[image]
The problem is that rays with more vertical source angles bounce off the bottom and go off the domain to the left. In the prt files, there is a message like:
` ray is outside the box where the ocean soundspeed is defined
  x = ( r, z ) =   -1027.3710183331586        25.478280757985310`
And no matter how much more distance I define on the left, it always fails a bit further than what I defined. As far as I am concerned, those rays are lost. Is there a way to tell Bellhop to stop tracing them?
I saw something about drawing a box out of which Bellhop doesn't care (http://oalib.hlsresearch.com/Rays/HLS-2010-1.pdf page 10), but didn't find anything about how to define it in arlpy and it sounds like it is from 0 to X range on the right, while my problem is to the left.
Thanks in advance!

mchitre, 2024-11-27:
`arlpy` supports a subset of all the options provided by BELLHOP based on common usage. If you find something that you need from BELLHOP but that isn't supported by `arlpy` yet, I am happy to take a PR that adds that option to `arlpy`.

Vanillaholic, 2025-01-08:
> Hello!  你好！
> 
> I'm trying to use python to simulate marine environments and compute rays, arrival times etc, as I am not familiar with matlab. All works fine ( I can set and plot the environment), however when I try to compute the rays or arrival times ( arrivals = pm.compute_arrivals(env) ) I get the following message in jupyter:我正在尝试使用 python 模拟海洋环境并计算光线、到达时间等，因为我对 matlab 不太熟悉。一切正常（我可以设置和绘制环境），但是当我尝试计算光线或到达时间（ arrivals = pm.compute_arrivals(env) ）时，我在 jupyter 中收到了以下消息：
> 
> [WARN] Bellhop did not generate expected output file.[WARN] Bellhop 未生成预期的输出文件。
> 
> ## As a result I can not plot the rays or arrival times, as the variable values are probably set to Nan or something. Specifically, when I try to plot arrival times I get this error:
> 结果是我无法绘制光线或到达时间，因为变量值可能被设置为 Nan 或其他。具体来说，当我试图绘制到达时间时，我得到了这个错误：
> AttributeError Traceback (most recent call last)AttributeError 回溯（最近的调用最后一次） Cell In[16], line 2单元格 In[16]，第 2 行 1 arrivals = pm.compute_arrivals(env) ----> 2 pm.plot_rays(arrivals, env=env, width=900)
> 
> File ~\AppData\Roaming\Python\Python311\site-packages\arlpy\uwapm.py:478, in plot_rays(rays, env, invert_colors, **kwargs)文件 ~\AppData\Roaming\Python\Python311\site-packages\arlpy\uwapm.py:478, 在 plot_rays(rays, env, invert_colors, **kwargs) 中 461 def plot_rays(rays, env=None, invert_colors=False, **kwargs): 462 """Plots ray paths.绘制光线路径。 463 464 :param rays: ray paths464 :param rays: 光线路径 (...) 476 >>> pm.plot_rays(rays, width=1000) 477 """ --> 478 rays = rays.sort_values('bottom_bounces', ascending=False) 479 max_amp = _np.max(_np.abs(rays.bottom_bounces)) if len(rays.bottom_bounces) > 0 else 0 480 if max_amp <= 0:如果 max_amp <= 0:
> 
> AttributeError: 'NoneType' object has no attribute 'sort_values'AttributeError: 'NoneType'对象没有属性'sort_values'
> 
> I do not know what the problem is and since there is not much information online about bellhop in python, I am stuck. Any help would be really appriciated. Thanks in advance!我不知道问题是什么，由于网上关于 Python 中 bellhop 的信息不多，我陷入了困境。任何帮助都将不胜感激。提前感谢！
> 
> Vassilis

have you solved the issue? I encountered the same problem as you

mchitre, 2025-04-15:
> I saw something about drawing a box out of which Bellhop stops tracing rays (http://oalib.hlsresearch.com/Rays/HLS-2010-1.pdf page 10), but didn't find anything about how to define it in arlpy. It is automatically set to 101% of the distance to the furthest receiver, and indeed defining an extra sound speed column at 102% of the distance, but backwards, fixed it. Would it be possible to define that value (and by extension its depth equivalent) in arlpy? The current values are good for defaults, but it would be good to be able to set them as well. Also, I assume that is more dependent on Bellhop, but it would be great to be able to define the box in -range, or just stop tracing rays that go into "negative" range. Thanks in advance!

I guess the concrete request here is to be able to define some bounding box parameters manually rather than the defaults that `arlpy` uses. I am OK with that, if we can specify exactly which parameters we would like to define. @Arthanor, if you can help define those, perhaps you can open a new issue with the details to track this?

This current issue description and title doesn't do justice to the request, so closing this.

closed 2025-04-15 by mchitre