org-arl/arlpy #106: Transmission Loss Plot Range Doesn't Match Expected Values (pm.plot_transmission_loss)

muhammadoginh, 2024-09-15:
Hello everyone,

I'm trying to plot transmission loss after placing the receiver in a grid, but the figure I get shows a range that doesn't match the actual values. The range should be from 0 to 1000.

Could anyone help me understand why this might be happening and how to fix it?

Thank you!
[image]

mchitre, 2024-09-15:
What you have provided isn't runnable code that one could run to reproduce your problem. I suggest you post a MWE (minimum working example) that allows your problem to be reproduced if you need help debugging.

Alternatively, you can debug this yourself by looking at the returned `tloss` to check what ranges it has computed transmission loss for. The code that plots simply takes the range to plot from what is in `tloss`.

muhammadoginh, 2024-09-15:
Thank you for your reply.

here is my simple code to reproduce the problem:
# import all required library
import arlpy.uwapm as pm
import arlpy.plot as plt
import numpy as np
import pandas as pd
import math

# Specify the bathymetry
bathy = [
    [0, 80],        # 80m water depth at the transmitter 
    [500, 80],      # 80m water depth 500m away
    [1000, 80]     # 80m water depth 1000m away
]

# Specify the sound speed profile (SSP)
ssp = [
    [0, 1503.0],   #1530 m/s at the surface
    [10, 1500.0],  #1500 m/s at 10m depth
    [15, 1498.0],  #1498 m/s at 15m depth
    [20, 1493.5],  # 1493.5 m/s at 20m depth
    [22, 1492.0],  #1492 m/s at 22m depth
    [23, 1490.0],  #1490 m/s at 23m depth
    [28, 1480.0],  #1480 m/s at 28m depth
    [30, 1477.0],  #1477 m/s at 30m depth
    [37, 1470.0],  #1470 m/s at 37m depth
    [40, 1468.5],  #1468.5 m/s at 40m depth
    [50, 1466.0],  #1466 m/s at 50m depth
    [60, 1464.5],  #1464.5 m/s at 60m depth
    [66, 1464.3],  #1464.3 m/s at 66m depth
    [70, 1464.0],  #1464 m/s at 70m depth
    [80, 1462.0]   #1462 m/s at 80m depth
]

# Create environment for no relay
env_no_relay = pm.create_env2d(
    frequency=10000,
    depth=bathy,
    soundspeed=ssp,
    soundspeed_interp ='linear',    # spline/linear
    bottom_soundspeed=1800,         # m/s
    bottom_density=2050,            # kg/m^3
    bottom_absorption=0.5,          # dB/wavelength
    bottom_roughness=0,             # m (rms)
    tx_depth=10,
    rx_depth=10,
    rx_range=1000,
    min_angle=-70,
    max_angle=70,
    nbeams=0                       # number of beams (0 = auto)
)

# for measurement purpose place a lots of receiver in a grid to visualize transmission loss
env_no_relay['rx_range'] = np.linspace(0, 1000, 1001)
env_no_relay['rx_depth'] = np.linspace(0, 80, 81)

# visualize transmission loss
tloss = pm.compute_transmission_loss(env_no_relay)
pm.plot_transmission_loss(tloss, env=env_no_relay, clim=[-60, -30], width=900)

when I plot transmission loss, I get the problem, I expect range (m) should be from 0 to 1000.

I also output the tloss, here it is
[image]

muhammadoginh, 2024-09-15:
Finally, I found a way. I changed a line in the plot_transmission_loss function, so I got the expected range (in this case, 0 to 1000). However, I still wonder why the columns in tloss don't show the actual values.
[image]

mchitre, 2024-09-15:
I would not recommend modifying the `env` after creation by changing fields. Try passing in the `rx_range` and `rx_depth` arrays when creating the environment in the first place.

If you still have a problem, you can run the `compute_transmission_loss` with `debug=True` argument and manually inspect the environment file passed in to BELLHOP to check.

muhammadoginh, 2024-09-15:
I have run compute_transmission_loss with the debug=True argument, and the output I received is:

```
'arlpy'
10000.000000
1
'CVWT'
1 0.0 80.000000
0.000000 1503.000000 /
10.000000 1500.000000 /
15.000000 1498.000000 /
20.000000 1493.500000 /
22.000000 1492.000000 /
23.000000 1490.000000 /
28.000000 1480.000000 /
30.000000 1477.000000 /
37.000000 1470.000000 /
40.000000 1468.500000 /
50.000000 1466.000000 /
60.000000 1464.500000 /
66.000000 1464.300000 /
70.000000 1464.000000 /
80.000000 1462.000000 /
'A*' 0.000000
80.000000 1800.000000 0.0 2.050000 0.500000 /
1
10.000000 /
81
0.000000 1.000000 2.000000 3.000000 4.000000 5.000000 6.000000 7.000000 8.000000 9.000000 10.000000 11.000000 12.000000 13.000000 14.000000 15.000000 16.000000 17.000000 18.000000 19.000000 20.000000 21.000000 22.000000 23.000000 24.000000 25.000000 26.000000 27.000000 28.000000 29.000000 30.000000 31.000000 32.000000 33.000000 34.000000 35.000000 36.000000 37.000000 38.000000 39.000000 40.000000 41.000000 42.000000 43.000000 44.000000 45.000000 46.000000 47.000000 48.000000 49.000000 50.000000 51.000000 52.000000 53.000000 54.000000 55.000000 56.000000 57.000000 58.000000 59.000000 60.000000 61.000000 62.000000 63.000000 64.000000 65.000000 66.000000 67.000000 68.000000 69.000000 70.000000 71.000000 72.000000 73.000000 74.000000 75.000000 76.000000 77.000000 78.000000 79.000000 80.000000 /
1001
0.000000 0.001000 0.002000 0.003000 0.004000 0.005000 0.006000 0.007000 0.008000 0.009000 0.010000 0.011000 0.012000 0.013000 0.014000 0.015000 0.016000 0.017000 0.018000 0.019000 0.020000 0.021000 0.022000 0.023000 0.024000 0.025000 0.026000 0.027000 0.028000 0.029000 0.030000 0.031000 0.032000 0.033000 0.034000 0.035000 0.036000 0.037000 0.038000 0.039000 0.040000 0.041000 0.042000 0.043000 0.044000 0.045000 0.046000 0.047000 0.048000 0.049000 0.050000 0.051000 0.052000 0.053000 0.054000 0.055000 0.056000 0.057000 0.058000 0.059000 0.060000 0.061000 0.062000 0.063000 0.064000 0.065000 0.066000 0.067000 0.068000 0.069000 0.070000 0.071000 0.072000 0.073000 0.074000 0.075000 0.076000 0.077000 0.078000 0.079000 0.080000 0.081000 0.082000 0.083000 0.084000 0.085000 0.086000 0.087000 0.088000 0.089000 0.090000 0.091000 0.092000 0.093000 0.094000 0.095000 0.096000 0.097000 0.098000 0.099000 0.100000 0.101000 0.102000 0.103000 0.104000 0.105000 0.106000 0.107000 0.108000 0.109000 0.110000 0.111000 0.112000 0.113000 0.114000 0.115000 0.116000 0.117000 0.118000 0.119000 0.120000 0.121000 0.122000 0.123000 0.124000 0.125000 0.126000 0.127000 0.128000 0.129000 0.130000 0.131000 0.132000 0.133000 0.134000 0.135000 0.136000 0.137000 0.138000 0.139000 0.140000 0.141000 0.142000 0.143000 0.144000 0.145000 0.146000 0.147000 0.148000 0.149000 0.150000 0.151000 0.152000 0.153000 0.154000 0.155000 0.156000 0.157000 0.158000 0.159000 0.160000 0.161000 0.162000 0.163000 0.164000 0.165000 0.166000 0.167000 0.168000 0.169000 0.170000 0.171000 0.172000 0.173000 0.174000 0.175000 0.176000 0.177000 0.178000 0.179000 0.180000 0.181000 0.182000 0.183000 0.184000 0.185000 0.186000 0.187000 0.188000 0.189000 0.190000 0.191000 0.192000 0.193000 0.194000 0.195000 0.196000 0.197000 0.198000 0.199000 0.200000 0.201000 0.202000 0.203000 0.204000 0.205000 0.206000 0.207000 0.208000 0.209000 0.210000 0.211000 0.212000 0.213000 0.214000 0.215000 0.216000 0.217000 0.218000 0.219000 0.220000 0.221000 0.222000 0.223000 0.224000 0.225000 0.226000 0.227000 0.228000 0.229000 0.230000 0.231000 0.232000 0.233000 0.234000 0.235000 0.236000 0.237000 0.238000 0.239000 0.240000 0.241000 0.242000 0.243000 0.244000 0.245000 0.246000 0.247000 0.248000 0.249000 0.250000 0.251000 0.252000 0.253000 0.254000 0.255000 0.256000 0.257000 0.258000 0.259000 0.260000 0.261000 0.262000 0.263000 0.264000 0.265000 0.266000 0.267000 0.268000 0.269000 0.270000 0.271000 0.272000 0.273000 0.274000 0.275000 0.276000 0.277000 0.278000 0.279000 0.280000 0.281000 0.282000 0.283000 0.284000 0.285000 0.286000 0.287000 0.288000 0.289000 0.290000 0.291000 0.292000 0.293000 0.294000 0.295000 0.296000 0.297000 0.298000 0.299000 0.300000 0.301000 0.302000 0.303000 0.304000 0.305000 0.306000 0.307000 0.308000 0.309000 0.310000 0.311000 0.312000 0.313000 0.314000 0.315000 0.316000 0.317000 0.318000 0.319000 0.320000 0.321000 0.322000 0.323000 0.324000 0.325000 0.326000 0.327000 0.328000 0.329000 0.330000 0.331000 0.332000 0.333000 0.334000 0.335000 0.336000 0.337000 0.338000 0.339000 0.340000 0.341000 0.342000 0.343000 0.344000 0.345000 0.346000 0.347000 0.348000 0.349000 0.350000 0.351000 0.352000 0.353000 0.354000 0.355000 0.356000 0.357000 0.358000 0.359000 0.360000 0.361000 0.362000 0.363000 0.364000 0.365000 0.366000 0.367000 0.368000 0.369000 0.370000 0.371000 0.372000 0.373000 0.374000 0.375000 0.376000 0.377000 0.378000 0.379000 0.380000 0.381000 0.382000 0.383000 0.384000 0.385000 0.386000 0.387000 0.388000 0.389000 0.390000 0.391000 0.392000 0.393000 0.394000 0.395000 0.396000 0.397000 0.398000 0.399000 0.400000 0.401000 0.402000 0.403000 0.404000 0.405000 0.406000 0.407000 0.408000 0.409000 0.410000 0.411000 0.412000 0.413000 0.414000 0.415000 0.416000 0.417000 0.418000 0.419000 0.420000 0.421000 0.422000 0.423000 0.424000 0.425000 0.426000 0.427000 0.428000 0.429000 0.430000 0.431000 0.432000 0.433000 0.434000 0.435000 0.436000 0.437000 0.438000 0.439000 0.440000 0.441000 0.442000 0.443000 0.444000 0.445000 0.446000 0.447000 0.448000 0.449000 0.450000 0.451000 0.452000 0.453000 0.454000 0.455000 0.456 [... cut ...]  0.952000 0.953000 0.954000 0.955000 0.956000 0.957000 0.958000 0.959000 0.960000 0.961000 0.962000 0.963000 0.964000 0.965000 0.966000 0.967000 0.968000 0.969000 0.970000 0.971000 0.972000 0.973000 0.974000 0.975000 0.976000 0.977000 0.978000 0.979000 0.980000 0.981000 0.982000 0.983000 0.984000 0.985000 0.986000 0.987000 0.988000 0.989000 0.990000 0.991000 0.992000 0.993000 0.994000 0.995000 0.996000 0.997000 0.998000 0.999000 1.000000 /
'C'
0
-70.000000 70.000000 /
0.0 80.800000 1.010000

```

mchitre, 2024-09-16:
The environment file seems OK. Could you post the `shd` file you get at the output here to check if that seems OK?

mchitre, 2024-10-12:
If I run your `env` file with my `bellhop`, the `shd` file I get is OK. If I use your `shd` file, the metadata in there seems to read incorrectly. I am running `BELLHOP 2022_4_20`, so I am guessing something changed in the `shd` file format in `BELLHOP 2023_05_18`. Could you try with an older version and see if it fixes the problem?

muhammadoginh, 2024-10-12:
> If I run your `env` file with my `bellhop`, the `shd` file I get is OK. If I use your `shd` file, the metadata in there seems to read incorrectly. I am running [`BELLHOP 2022_4_20`](http://oalib.hlsresearch.com/AcousticsToolbox/at_2022_4_20.zip), so I am guessing something changed in the `shd` file format in [`BELLHOP 2023_5_18`](http://oalib.hlsresearch.com/AcousticsToolbox/at_2022_5_18.zip). Could you try with `BELLHOP 2022_4_20` and see if it fixes the problem?

Ahh, problem solved, you're right, thank you so much!

closed 2024-10-14 by muhammadoginh