org-arl/arlpy #109: samples run failed

Vanillaholic, 2025-01-08:
hi , sorry to bother you . when I ran samples/bellhop.ipynb,  it displayed issue below. Do you know how to solve this ? Looking forward to your reply!

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[10], line 2
      1 rays = pm.compute_eigenrays(env)
----> 2 pm.plot_rays(rays, env=env, width=900)

File d:\anaconda3\.conda\envs\torch_env\lib\site-packages\arlpy\uwapm.py:484, in plot_rays(rays, env, invert_colors, **kwargs)
    467 def plot_rays(rays, env=None, invert_colors=False, **kwargs):
    468     """Plots ray paths.
    469 
    470     :param rays: ray paths
   (...)
    482     >>> pm.plot_rays(rays, width=1000)
    483     """
--> 484     rays = rays.sort_values('bottom_bounces', ascending=False)
    485     max_amp = _np.max(_np.abs(rays.bottom_bounces)) if len(rays.bottom_bounces) > 0 else 0
    486     if max_amp <= 0:

AttributeError: 'NoneType' object has no attribute 'sort_values'

Vanillaholic, 2025-01-08:
[image]
This is place where went wrong

mchitre, 2025-01-08:
Run your code with debug=True and copy out the generated files. Looking at their content might give you a hint as to what is wrong.

Vanillaholic, 2025-01-08:
sorry to bother you again，it seems I set the path wrong， do you have docs or webs about setting the bellhop.exe program path？

mchitre, 2025-01-09:
Path setting depends on OS, shell, etc. Generally you need an environment variable PATH to be setup to include the folder where bellhop.exe resides. If this is correctly setup, you should be able to type `bellhop.exe` in a shell successfully before launching Jupyter notebook.

mchitre, 2025-02-09:
@Vanillaholic is this resolved?

Vanillaholic, 2025-02-10:
The issue has been resolved. There was an error with the binary files in the acoustic toolbox I downloaded. Thank you for your help.

closed 2025-02-10 by mchitre