org-arl/arlpy #110: Bellhop did not generate expected output file

zwc1826, 2025-01-10:
Sorry to bother you. When running the sample, the following issue occurs, how can i fix it?

import arlpy.uwapm as pm
import arlpy.plot as plt
import numpy as np

env = pm.create_env2d()
rays = pm.compute_eigenrays(env, debug=True)

[DEBUG] Model: bellhop
[WARN] Bellhop did not generate expected output file
[DEBUG] Bellhop working files: C:\Users\zwc\AppData\Local\Temp\tmphz3lu7ey.*``

mchitre, 2025-01-10:
Check the `prt` file in the folder to see what the error is.

zwc1826, 2025-01-10:
There is no prt file in the folder.

mchitre, 2025-01-10:
There will be a `log` file. Check its content.

The output tells you that you have a debug folder with files telling you what failed. Look at the contents of those files as they will give you clues as to what went wrong.

zwc1826, 2025-01-10:
I checked the folders carefully, but there is no log file. And amazing, the bellhop.ipynb runs successfully, but when the same codes are run in .py, the above issue occurs...

mchitre, 2025-01-10:
That sounds like your `PATH` is not properly setup to find `bellhop.exe`

zwc1826, 2025-01-10:
Thank you very much! That is because the Pycharm was open before the PATH is setup, so i restart Pycharm, and the issue vanishes!

mchitre, 2025-01-10:
Great!

closed 2025-01-10 by mchitre