org-arl/arlpy #114: index and columns reading format may be incorrect (Bellhop from AT 2023 updated version)

xuliang5115, 2025-02-17:
import arlpy.uwapm as pm
import numpy as np

env = pm.create_env2d(
    frequency=10000,
    soundspeed=[
        [0, 1540],  # 1540 m/s at the surface
        [10, 1530],  # 1530 m/s at 10 m depth
        [20, 1532],  # 1532 m/s at 20 m depth
        [25, 1533],  # 1533 m/s at 25 m depth
        [30, 1535]  # 1535 m/s at the seabed
    ],
    bottom_soundspeed=1550,
    bottom_density=1200,
    bottom_absorption=1.0,
    tx_depth=15,
    depth=[
        [0, 25],  # 30 m water depth at the transmitter
        [300, 30],  # 20 m water depth 300 m away
        [1000, 20]  # 25 m water depth at 1 km
    ]
)

env['rx_range'] = np.linspace(0, 1000, 1001)
env['rx_depth'] = np.linspace(0, 30, 301)

tloss = pm.compute_transmission_loss(env)
print(tloss.shape)  
xr = (min(tloss.columns), max(tloss.columns))
print(xr[0], xr[1])
pm.plot_transmission_loss(tloss, env=env, clim=[-60, -30], width=900)

[image]

xuliang5115, 2025-02-17:
Then some path testing may be needed, as the current code seems to require Bellhop.exe and .py to be in the same folder, otherwise the output file cannot be detected, or move from the path of __file__ to the path where .exe is located during calculation and plotting.

mchitre, 2025-02-17:
There is no special requirement of where `bellhop.exe` is, as long as it in on your `PATH`.

mchitre, 2025-02-17:
It's not clear to me from the issue description as to what the problem is. The code and screenshot doesn't describe anything. Please update the description to explain what the problem is and how to reproduce it.

mchitre, 2025-04-15:
Closing since nothing is actionable given the information.

closed 2025-04-15 by mchitre