org-arl/arlpy #115: pip install arlpy fails while compiling scipy

almaniac, 2025-02-19:
I'm trying to use arlpy for the first time.  When I try to install arlpy I get an error.  It looks to me like pip is looking for a Fortran compiler while installing scipy and not finding it.  I haven't seen anything in the documentation to help solve the issue.

I've tried installing scipy in my virtual environment before installing arlpy, but that did not solve the problem.

mchitre, 2025-02-19:
If is impossible to suggest solutions without knowing the exact error. If the error is in installing `scipy`, you may be better off asking on StackOverflow or Python forums. `arlpy` does not directly use FORTRAN, so it's unlikely that a FORTRAN error comes from it.

almaniac, 2025-02-19:
It is definitely a scipy problem.  It seems to be isolated to version 1.13.0 of scipy that pip tries to install with arlpy.

almaniac, 2025-02-21:
I was able to get it working by downloading the arlpy code and creating my own environment to run it in.  I installed numpy==1.26.4 and then installled scipy, pandas, utm, bokeh, and matplot lib without specifying the version.  

So far, so good.