org-arl/arlpy #118: Documentation: Seabed property units

JPalmerK, 2025-05-26:
Hey there,
Thanks for putting this package together. In the documentation it would be great to have some confirmation on the units. Specifically for the density. I believe in the original model they are kg/m^3 but I've noticed the default value is 1600 which is the same as the default sound speed for the bottom. It's unclear whether thats just a carryover or the system is looking for different units. Either way, having all of the units made explicit would be very helpful. Thanks!

mchitre, 2025-05-27:
Yes, the unit is kg/m³. Why do you feel that 1600 kg/m³ is not reasonable or confusing?

Happy to take in a documentation update PR, if you'd like to contribute one.

mchitre, 2025-05-28:
Yes, bellhop uses non SI units in some places. In arlpy, I covert SI units to bellhop ones where necessary. So in arlpy, you specify SI units.