org-arl/arlpy #119: Transmission Loss x-axis isn't right

Vanillaholic, 2025-07-02:
Hi, when plotting transmission loss in a long range scenario, the x--axis went wrong
'''python
env['rx_range'] = np.linspace(0, 200000, 1001)
env['rx_depth'] = np.linspace(0, 5000, 101)

tloss = pm.compute_transmission_loss(env)
pm.plot_transmission_loss(tloss,env=env,width=900)
''' 
the attribute of tloss--columns isn't correponded  to the figure ,which leads to a wrong x-coordinate.
[image]

should fix the function `plot_transmission_loss()` in file **_uwapm.py_** , 

'''python
def plot_transmission_loss(tloss,env, **kwargs):
    if env is None:
        raise ValueError("Environment is required")
    xr = (env['rx_range'].min(), env['rx_range'].max())
    yr = (-max(tloss.index), -min(tloss.index))
    xlabel = 'Range (m)'
    if xr[1]-xr[0] > 10000:
        xr = (env['rx_range'].min()/1000, env['rx_range'].max()/1000)
        xlabel = 'Range (km)'
    oh = _plt.hold()
    _plt.image(20*_np.log10(_fi.epsilon+_np.abs(_np.flipud(_np.array(tloss)))), x=xr, y=yr, xlabel=xlabel, ylabel='Depth (m)', xlim=xr, ylim=yr, **kwargs)
    _plt.hold(oh)
'''

the correct result is below
[image]

all the best

mchitre, 2025-07-03:
Thanks! Could you raise a PR with the proposed fix please?

Vanillaholic, 2025-07-03:
PR has submitted